ThisIsNotRocketScience/GerberTools #3: line widths

luftek, 2016-11-30:
Sorry, me again :) 👍 line widths are messed sometimes. By deleting instances of Gerbers and placing them again something changes. Sorry seems it is not 100% reproducable.

Any changes in xml format from last versions?
How could number of instances affect line width? (not including "your" tabs)
Photos before - max. number of instances 4
Last photo is OK - max. number of instances 3

[image]

[image]
[image]
[image]

StijnKuipers, 2016-12-02:
Usually this means the final command in the previous gerber leaves something open - or a state is not correctly reset - Merging this state will then propagate on... or maybe the initial aperture is not selected correctly.  Could you mail me these 3 gerbersets so I can prod them around? - this seems like a nicely isolated testcase. 

I will also make an easier-to-copy aboutbox.

luftek, 2016-12-03:
it is 6 gerbers and gerberset.
gerberset will need a change if you want to use (fixed folder paths)
https://github.com/luftek/LimeSDR-USB/tree/master/hardware/plug/1v2/Enclosure/0v6_76x35/panel/_Gerber/20161104_combined 

download https://github.com/luftek/LimeSDR-USB/tree/master/hardware/plug/1v2/Enclosure 
with: https://minhaskamal.github.io/DownGit/#/home

StijnKuipers, 2016-12-03:
Thankyou! - will dive in to this tonight.
I'm aware of the relative path issue - I'll give that some attention soon as well

The aboutbox has been fixed btw

luftek, 2016-12-03:
I watch the progress, have an app on phone :) thank you for this

StijnKuipers, 2016-12-03:
Ok - turned out to be something else. There was a special path for "massive merges" that wasnt tested very well - in the cases I tried (many instances of the same board) - this was no issue - but for 8+ different board designs some of the apertures got mixed up. Should be correct now.

closed 2016-12-03 by StijnKuipers

luftek, 2016-12-03:
Awesome, that you found it

StijnKuipers, 2016-12-03:
Build updated on thisisnotrocketscience.nl!