ThisIsNotRocketScience/GerberTools #5: Struggling to build

TomKeddie, 2017-01-12:
Sorry to raise this as an issue but i couldn't find an email address for you.

I'm using Visual Studio Community 2015, 14.0.25431.01 and see these errors trying to build the library at head as of now.

```
1>Z:\tom\git\ThisIsNotRocketScience\GerberTools\GerberLibrary\Artwork Related\SVGWriter.cs(126,81,126,86): error CS1525: Invalid expression term 'throw'
1>Z:\tom\git\ThisIsNotRocketScience\GerberTools\GerberLibrary\Artwork Related\SVGWriter.cs(126,81,126,86): error CS1002: ; expected
1>Z:\tom\git\ThisIsNotRocketScience\GerberTools\GerberLibrary\Artwork Related\SVGWriter.cs(127,9,127,15): error CS1014: A get or set accessor expected
1>Z:\tom\git\ThisIsNotRocketScience\GerberTools\GerberLibrary\Artwork Related\SVGWriter.cs(127,15,127,16): error CS1014: A get or set accessor expected
1>Z:\tom\git\ThisIsNotRocketScience\GerberTools\GerberLibrary\Artwork Related\SVGWriter.cs(127,16,127,23): error CS1014: A get or set accessor expected
1>Z:\tom\git\ThisIsNotRocketScience\GerberTools\GerberLibrary\Artwork Related\SVGWriter.cs(127,23,127,24): error CS1014: A get or set accessor expected
1>Z:\tom\git\ThisIsNotRocketScience\GerberTools\GerberLibrary\Artwork Related\SVGWriter.cs(127,24,127,33): error CS1014: A get or set accessor expected
1>Z:\tom\git\ThisIsNotRocketScience\GerberTools\GerberLibrary\Artwork Related\SVGWriter.cs(127,33,127,34): error CS1014: A get or set accessor expected
1>Z:\tom\git\ThisIsNotRocketScience\GerberTools\GerberLibrary\Artwork Related\SVGWriter.cs(127,34,127,51): error CS1014: A get or set accessor expected
1>Z:\tom\git\ThisIsNotRocketScience\GerberTools\GerberLibrary\Artwork Related\SVGWriter.cs(127,52,127,69): error CS1014: A get or set accessor expected
1>Z:\tom\git\ThisIsNotRocketScience\GerberTools\GerberLibrary\Artwork Related\SVGWriter.cs(127,69,127,70): error CS1014: A get or set accessor expected
1>Z:\tom\git\ThisIsNotRocketScience\GerberTools\GerberLibrary\Artwork Related\SVGWriter.cs(127,70,127,87): error CS1014: A get or set accessor expected
1>Z:\tom\git\ThisIsNotRocketScience\GerberTools\GerberLibrary\Artwork Related\SVGWriter.cs(127,97,127,102): error CS1525: Invalid expression term 'throw'
1>Z:\tom\git\ThisIsNotRocketScience\GerberTools\GerberLibrary\Artwork Related\SVGWriter.cs(127,97,127,102): error CS1002: ; expected
1>Z:\tom\git\ThisIsNotRocketScience\GerberTools\GerberLibrary\Artwork Related\SVGWriter.cs(127,141,127,146): error CS1525: Invalid expression term 'throw'
1>Z:\tom\git\ThisIsNotRocketScience\GerberTools\GerberLibrary\Artwork Related\SVGWriter.cs(127,141,127,146): error CS1002: ; expected
1>Z:\tom\git\ThisIsNotRocketScience\GerberTools\GerberLibrary\Artwork Related\SVGWriter.cs(127,179,127,179): error CS1513: } expected
1>Z:\tom\git\ThisIsNotRocketScience\GerberTools\GerberLibrary\Core\GraphicsInterface.cs(168,34,168,36): error CS1043: { or ; expected
1>Z:\tom\git\ThisIsNotRocketScience\GerberTools\GerberLibrary\Core\GraphicsInterface.cs(168,34,168,36): error CS1513: } expected
1>Z:\tom\git\ThisIsNotRocketScience\GerberTools\GerberLibrary\Core\GraphicsInterface.cs(168,81,168,86): error CS1525: Invalid expression term 'throw'
1>Z:\tom\git\ThisIsNotRocketScience\GerberTools\GerberLibrary\Core\GraphicsInterface.cs(168,81,168,86): error CS1002: ; expected
1>Z:\tom\git\ThisIsNotRocketScience\GerberTools\GerberLibrary\Core\GraphicsInterface.cs(168,119,168,119): error CS1513: } expected
```

```
 126:       public bool Dotted { get => throw new NotImplementedException(); set => throw new NotImplementedException(); }
127:        System.Drawing.Drawing2D.InterpolationMode GraphicsInterface.InterpolationMode { get => throw new NotImplementedException(); set => throw new NotImplementedException(); }

```

TomKeddie, 2017-01-12:
This is interesting, seems you're using a newer version?

```
> diff --git a/GerberProjects/GerberProjects.sln b/GerberProjects/GerberProjects.sln
> index 8ce5b7b..9715731 100644
> --- a/GerberProjects/GerberProjects.sln
> +++ b/GerberProjects/GerberProjects.sln
> @@ -1,7 +1,7 @@
>  ï»¿
>  Microsoft Visual Studio Solution File, Format Version 12.00
> -# Visual Studio 15
> -VisualStudioVersion = 15.0.26020.0
> +# Visual Studio 14
> +VisualStudioVersion = 14.0.25420.1
> 
```

StijnKuipers, 2017-01-12:
could be, upgraded everything to the latest RC's recently.. but those errors should still not be there - that code is older.. Lets see what happened to those 2 files

- after some investigation, seems that the automated implementation for missing-interface implementation-functions has upgraded to a new throwing style - I'll just add them for you :-)

StijnKuipers, 2017-01-12:
Ok - committed some fixes for those - let me know if this works :-)

TomKeddie, 2017-01-13:
Thanks, that builds well now.  Looking forward to smoother panelising.

closed 2017-01-13 by TomKeddie