ThisIsNotRocketScience/GerberTools #7: New version, rendering: Panelizer vs. QuickRender

luftek, 2017-01-21:
http://blog.thisisnotrocketscience.nl/wp-content/uploads/2015/02/2017-01-17-PanelTools.zip 
[image]

StijnKuipers, 2017-01-23:
Ive noticed a few odd regressions with the renderer called from the Panelizer - could you send me this file to test?

luftek, 2017-01-23:
[gerbers.zip](https://github.com/ThisIsNotRocketScience/GerberTools/files/724521/gerbers.zip)
Change file paths in *.gerberset

StijnKuipers, 2017-01-26:
Fixed! Rendergeometry scaling got hit by a premature optimization -> the circles ended up with 0 vertices due to a far too low geometry setting with no sanity checking. 

Will upload new build today!

closed 2017-01-26 by StijnKuipers

StijnKuipers, 2017-01-27:
Added a new GitHub release! From now on the updates shall be packaged here

luftek, 2017-01-27:
Thank you! Managing releases here is awesome 👍