ThisIsNotRocketScience/GerberTools #8: Breaktab artefact

luftek, 2017-01-30:
Release: c4c91e2 2017-01-27
Some artefact of a breaktab.
Small offset of 0.5mm or changed dimension of breaktab solves the issue

[image]
[image]

Demo:
[demo.zip](https://github.com/ThisIsNotRocketScience/GerberTools/files/740548/demo.zip)

StijnKuipers, 2017-02-05:
Will take a look tonight! Nice catch.

thatch, 2017-04-07:
I saw the same behavior on a horizontal gap as well with the latest released version (2017-02-17).