ThisIsNotRocketScience/GerberTools #9: Panelizer: how to open KiCad gerbers

rusefi, 2017-02-06:
I have a folder with my gerbers geberated by KiCad - for example uncompressed https://svn.code.sf.net/p/rusefi/code/trunk/hardware/brain_board/gerber/brain_board_STM32F407_R0.2.zip

I launch GerberTools_2017_1_27\Panelizer\GerberPanelizer_VS2015.exe and then what?

I do not have a .gerberset file - how do I create one?

I try opening my gerbers one by one but nothing is rendered. Please help.

luftek, 2017-02-06:
Did you change .gm1 outlinr file to .gko?

rusefi, 2017-02-06:
I have not changed .gm1 outlinr file to .gko, I did not know I needed to?

Assuming I should have renamed it, I have renamed my .gm1 to .gko but nothing seems to have changed - still cannot open individual files, still no idea about gerbersets

if that would help I am trying to open https://svn.code.sf.net/p/rusefi/code/trunk/hardware/rusefi_hardware_history/brain_board_SD_Card_1_rev00.10/gerber/brain_board_SD_Card_1_R0.01.zip (an uncompressed version of that file)

luftek, 2017-02-06:
Ok, you renamed .gm1 to .gko

Then:
1. Open Panelizer
2. Edit -> New
3. Board Placement -> Add Gerber Folder
4. For multiple instances of same gerber file click on instance in the right tree
5. Probably (0,0) coordinates (auxiliary axis) is not assigned on board edge so use: Board Placement -> Autopack:Naive

Other things are up to you to discover ;)

[image]

luftek, 2017-02-06:
About the auxiliary axis for easier board placement in the future:
[image]
[image]
[image]

StijnKuipers, 2017-02-07:
The concept is to add entire folders in 1 go - you cannot add individual gerber files.

Process is roughly:

1) start the .exe
2) drag the folders you want to the main area
3) a new panel document will start with 1 instance for each folder you dragged on to the area.
4) you can add more instances (rightclick menu) / save the current setup / move stuff around
5) export merged.

There are some notes in the included documentation - the layer-names roughly follow the defacto standards (see OSHPark, Seeed, PCBWay or DirtyPCBs etc)

closed 2017-02-07 by StijnKuipers