ThisIsNotRocketScience/GerberTools #10: Merge error with rounded corners, creates a diagonal

TomKeddie, 2017-02-20:
As per the attached files, the merged gerber has an error on two of the rounded corners.

[image]

[image]

[Grove Base Cape for BeagleBone v2.0-gerbers.zip](https://github.com/ThisIsNotRocketScience/GerberTools/files/786613/Grove.Base.Cape.for.BeagleBone.v2.0-gerbers.zip)

[panel-two.zip](https://github.com/ThisIsNotRocketScience/GerberTools/files/786614/panel-two.zip)

StijnKuipers, 2017-02-20:
thanks for the report! Will look in to this!

StijnKuipers, 2017-02-20:
May I ask if you tried this on the latest release? Or on an older build? The beaglecape seems to be fine when merged - so possibly the combination of the boards is an issue - will look further in to it. 

[image]

StijnKuipers, 2017-02-20:
Oh wait - I see 1 corner going wrong there anyway - but a different one :-D 
Yeeeey challenge ahead

TomKeddie, 2017-02-20:
Hi Stijn, am using this version, apologies for not including that information.  Cheers, Tom

Gerber Panelizer
Version 1.0.6218.28812 built on 1/13/2017 2:33:47 PM
Copyright ©  2015
This Is Not Rocket Science
A tool to combine several gerber files in to a single larger board

TomKeddie, 2017-02-20:
Ok, I tried the latest and get the same result as you, the issue with the larger corner is gone but the one with the smaller corner is still there.

PNL
Version 1.0.6257.24941 built on 2/17/2017 4:51:22 AM
Copyright ©  2015
This Is Not Rocket Science
A tool to combine several gerber files in to a single larger board