ThisIsNotRocketScience/GerberTools #14: Problems running under Linux mono

thatch, 2017-03-17:
It's my first time trying to run the tools, and I uncovered an error when running in the obvious way on Linux.  This occurs under both older Ubuntu (mono 3.2.8) and recent Arch (mono 4.8.0).  The same release works fine under Windows 10.  I haven't looked into whether previous GerberTools releases behave any better though.

```
$ cd GerberTools_2017_2_17/Panelizer
$ mono GerberPanelizer.exe

Unhandled Exception:
System.Reflection.TargetInvocationException: Exception has been thrown by the target of an invocation. ---> System.ArgumentException: XOR data length expected 2046820352, read 1370
Parameter name: stream
  at System.Drawing.Icon.InitFromStreamWithSize (System.IO.Stream stream, Int32 width, Int32 height) [0x00000] in <filename unknown>:0 
  at System.Drawing.Icon..ctor (System.Runtime.Serialization.SerializationInfo info, StreamingContext context) [0x00000] in <filename unknown>:0 
  at (wrapper managed-to-native) System.Reflection.MonoCMethod:InternalInvoke (System.Reflection.MonoCMethod,object,object[],System.Exception&)
  at System.Reflection.MonoCMethod.InternalInvoke (System.Object obj, System.Object[] parameters) [0x00000] in <filename unknown>:0 
  --- End of inner exception stack trace ---
  at System.Reflection.MonoCMethod.InternalInvoke (System.Object obj, System.Object[] parameters) [0x00000] in <filename unknown>:0 
  at System.Reflection.MonoCMethod.DoInvoke (System.Object obj, BindingFlags invokeAttr, System.Reflection.Binder binder, System.Object[] parameters, System.Globalization.CultureInfo culture) [0x00000] in <filename unknown>:0 
  at System.Reflection.MonoCMethod.Invoke (System.Object obj, BindingFlags invokeAttr, System.Reflection.Binder binder, System.Object[] parameters, System.Globalization.CultureInfo culture) [0x00000] in <filename unknown>:0 
  at System.Reflection.MethodBase.Invoke (System.Object obj, System.Object[] parameters) [0x00000] in <filename unknown>:0 
  at System.Runtime.Serialization.ObjectRecord.LoadData (System.Runtime.Serialization.ObjectManager manager, ISurrogateSelector selector, StreamingContext context) [0x00000] in <filename unknown>:0 
  at System.Runtime.Serialization.ObjectManager.DoFixups () [0x00000] in <filename unknown>:0 
  at System.Runtime.Serialization.Formatters.Binary.ObjectReader.ReadNextObject (System.IO.BinaryReader reader) [0x00000] in <filename unknown>:0 
  at System.Runtime.Serialization.Formatters.Binary.ObjectReader.ReadObjectGraph (BinaryElement elem, System.IO.BinaryReader reader, Boolean readHeaders, System.Object& result, System.Runtime.Remoting.Messaging.Header[]& headers) [0x00000] in <filename unknown>:0 
  at System.Runtime.Serialization.Formatters.Binary.BinaryFormatter.NoCheckDeserialize (System.IO.Stream serializationStream, System.Runtime.Remoting.Messaging.HeaderHandler handler) [0x00000] in <filename unknown>:0 
  at System.Runtime.Serialization.Formatters.Binary.BinaryFormatter.Deserialize (System.IO.Stream serializationStream) [0x00000] in <filename unknown>:0 
  at System.Resources.ResourceReader.ReadNonPredefinedValue (System.Type exp_type) [0x00000] in <filename unknown>:0 
  at System.Resources.ResourceReader.ReadValueVer2 (Int32 type_index) [0x00000] in <filename unknown>:0 
  at System.Resources.ResourceReader.LoadResourceValues (ResourceCacheItem[] store) [0x00000] in <filename unknown>:0 
  at System.Resources.ResourceReader+ResourceEnumerator.FillCache () [0x00000] in <filename unknown>:0 
  at System.Resources.ResourceReader+ResourceEnumerator..ctor (System.Resources.ResourceReader readerToEnumerate) [0x00000] in <filename unknown>:0 
  at System.Resources.ResourceReader.GetEnumerator () [0x00000] in <filename unknown>:0 
  at System.Resources.ResourceSet.ReadResources () [0x00000] in <filename unknown>:0 
  at System.Resources.ResourceSet.GetObjectInternal (System.String name, Boolean ignoreCase) [0x00000] in <filename unknown>:0 
  at System.Resources.ResourceSet.GetObject (System.String name, Boolean ignoreCase) [0x00000] in <filename unknown>:0 
  at System.Resources.RuntimeResourceSet.GetObject (System.String name, Boolean ignoreCase) [0x00000] in <filename unknown>:0 
  at System.Resources.ResourceManager.GetObject (System.String name, System.Globalization.CultureInfo culture) [0x00000] in <filename unknown>:0 
  at System.Resources.ResourceManager.GetObject (System.String name) [0x00000] in <filename unknown>:0 
  at GerberCombinerBuilder.GerberPanelizerParent.InitializeComponent () [0x00000] in <filename unknown>:0 
  at GerberCombinerBuilder.GerberPanelizerParent..ctor () [0x00000] in <filename unknown>:0 
  at (wrapper remoting-invoke-with-check) GerberCombinerBuilder.GerberPanelizerParent:.ctor ()
  at GerberCombinerBuilder.Program.Main () [0x00000] in <filename unknown>:0 
```

thatch, 2017-03-17:
I read the note at http://blog.thisisnotrocketscience.nl/projects/pcb-panelizer/ that says
1. command line tools should work, which I did not check.
2. that I'd need to "replace the openTK dll with the mono version" but unsure what that entails, as I'm not *that* familiar with Mono.

(The readme says "should work" without restriction though, so I tried the GUI first.)

I took a look at the source and this is related to loading an ico file.  I'm not sure which yet.

thatch, 2017-03-18:
The last line when running strce is an attempt to open GerberPanelizer.resources.dll (ENOENT), which makes me think it's attempting to load a PE resources from itself.  It appears that the icons in the release zip are all a format with valid headers, but then the entries all start with "\x89PNG".  I don't see code to load this in https://github.com/mono/mono/blob/master/mcs/class/System.Drawing/System.Drawing/Icon.cs (but there's a patch on https://bugzilla.xamarin.com/show_bug.cgi?id=30715 that does).

I'll try swapping out the icon resources and running again; I assume the normal build requires Visual Studio, and there isn't a commandline build script that works on Linux?

thatch, 2017-03-18:
Replacing it with Resource Hacker was insufficient to get it running under mono -- I'm not sure which .  I checked and the command line utilities do work, so perhaps the readme should just be adjusted to be more specific in the meantime?

StijnKuipers, 2017-03-18:
I'll test it on a linux vm - see if I can fix this somehow. If it really only is the PNG stuff it should be a relatively simple fix.  Did you try running mono with that patch? Very odd that png streams are not supported in this particular way. 

So far I've only built it on Visual Studio - if somebody wants to make/maintain a buildscript for other systems - be my guest. I will personally not make/maintain one - another support headache.

thatch, 2017-04-06:
With the patch it gets slightly further,

```
Unhandled Exception:
System.InvalidCastException: Specified cast is not valid.
  at System.Drawing.Icon.SaveBestSingleIcon (System.IO.BinaryWriter writer, System.Int32 width, System.Int32 height) [0x000b0] in <77ff46021c554b57be5b7a306d989e5d>:0 
  at System.Drawing.Icon.Save (System.IO.Stream outputStream, System.Int32 width, System.Int32 height) [0x00031] in <77ff46021c554b57be5b7a306d989e5d>:0 
  at System.Drawing.Icon.GetInternalBitmap () [0x00029] in <77ff46021c554b57be5b7a306d989e5d>:0 
  at System.Drawing.Icon.ToBitmap () [0x0001b] in <77ff46021c554b57be5b7a306d989e5d>:0 
  at (wrapper remoting-invoke-with-check) System.Drawing.Icon:ToBitmap ()
  at System.Windows.Forms.XplatUIX11.SetIcon (System.Windows.Forms.Hwnd hwnd, System.Drawing.Icon icon) [0x00021] in <e37081a3704b4473a59cdf5048ad96bc>:0 

```

I then hacked up GetInternalBitmap to do

```
if (imageData[0] is PNGImage) {
  bitmap = ((PNGImage)imageData[0]).png;
} else ...
```

and it gets further.  It launches and has a valid icon, but then when using File->New it crashes in OpenTK saying "The specified window isn't an OpenGL window."  Since you mentioned replacing this with the "mono version", I just built OpenTK from AUR and copied it in place... now New causes the window to become painted with the null brush.  Did you have something more specific in mind for "replacing"?

dromer, 2017-07-27:
Hmm, also running into this while trying to open this project with mono.

This machine happens to also have an nvidia gpu (and using the associated nvidia driver), but the solution to use __GL_THREADED_OPTIMIZATIONS=0 doesn't seem to work as I get the exact same System.Reflection.TargetInvocationException

Any ideas?

thatch, 2017-09-20:
@dromer The TargetInvocationException implies you need to remove openTK.dll from the local directory, so it uses the system-installed one.  (If you're on Arch, it's in AUR.)

dromer, 2017-09-20:
Hi @thatch ,  I renamed OpenTK.dll and installed libopentk1.1-cil (wich provides OpenTK.dll on ubuntu 16.04) and still get System.Reflection.TargetInvocationException

thatch, 2017-09-20:
Ah, I think I was incorrect then.  Does your error have "XOR data length expected" as my paste above?  If so, that's the PNG icon issue.  Ways forward are either:

1. Recompile with "normal" icon resources (requires Windows, basically)
2. Patch and recompile Mono.  There's a patch that worked well enough, but it doesn't apply to current mono-git anymore, and I need to get it fixed up.

dromer, 2017-09-20:
Yes I did get the 'XOR data length' line as with you.

I didn't try to compile the code myself, but used the GerberTools_2017_2_17.zip binary release.
Honestly I'm also not looking to patch and recompile mono for this.

phl0, 2017-11-13:
Is there any update to this? I am also failing the XOR stuff but would like to run the Panelizer on Linux.

davidbitton, 2017-11-13:
Same here compiling the latest from the repo using Visual Studio Mac.

runesoeknudsen, 2017-11-14:
I have looked at bit into the problem and created a pull request that fixes the problem, see  #41 
Basically I have extracted the icon file from the code and saved it as a separated file. Thanks for all the hints.

To make it easy for others I have here attached a compiled version: [Panelizer.zip](https://github.com/ThisIsNotRocketScience/GerberTools/files/1472105/Panelizer.zip)
This one I have succesfully used to panellized a PCB with Mono JIT compiler version 5.4.0.201

phl0, 2017-11-15:
Confirmed. Launched @runesoeknudsen s compiled version successfully on my Ubuntu 16.04 Box with mono 5.8.0.22

davidbitton, 2017-11-15:
@phl0 how are you launching it? i tried w/ `mono` from a cmd prompt on my Mac and it dumped a bunch of errors.

```
bittond:Panelizer/ $ mono GerberPanelizer.exe                                                                                                                             [11:32:12]
WARNING: The Carbon driver has not been ported to 64bits, and very few parts of Windows.Forms will work properly, or at all
Stacktrace:

  at <unknown> <0xffffffff>
  at (wrapper managed-to-native) System.Windows.Forms.XplatUICarbon.CGDisplayBounds (intptr) [0x00002] in <36fbc38531724d82b10aef6ebafea6e9>:0
  at System.Windows.Forms.XplatUICarbon.get_WorkingArea () [0x00005] in <36fbc38531724d82b10aef6ebafea6e9>:0
  at System.Windows.Forms.XplatUICarbon.get_VirtualScreen () [0x00000] in <36fbc38531724d82b10aef6ebafea6e9>:0
  at System.Windows.Forms.XplatUI.get_VirtualScreen () [0x00000] in <36fbc38531724d82b10aef6ebafea6e9>:0
  at System.Windows.Forms.Screen..cctor () [0x00034] in <36fbc38531724d82b10aef6ebafea6e9>:0
  at (wrapper runtime-invoke) object.runtime_invoke_void (object,intptr,intptr,intptr) [0x0001e] in <4fdc5ed61a074cafb49fa42deb20d521>:0
  at <unknown> <0xffffffff>
  at System.Windows.Forms.Hwnd.GetNextStackedFormLocation (System.Windows.Forms.CreateParams,System.Windows.Forms.Hwnd) [0x00064] in <36fbc38531724d82b10aef6ebafea6e9>:0
  at System.Windows.Forms.XplatUICarbon.CreateWindow (System.Windows.Forms.CreateParams) [0x000c5] in <36fbc38531724d82b10aef6ebafea6e9>:0
  at System.Windows.Forms.XplatUI.CreateWindow (System.Windows.Forms.CreateParams) [0x00000] in <36fbc38531724d82b10aef6ebafea6e9>:0
  at System.Windows.Forms.NativeWindow.CreateHandle (System.Windows.Forms.CreateParams) [0x00009] in <36fbc38531724d82b10aef6ebafea6e9>:0
  at System.Windows.Forms.Control.CreateHandle () [0x00031] in <36fbc38531724d82b10aef6ebafea6e9>:0
  at System.Windows.Forms.Form.CreateHandle () [0x00000] in <36fbc38531724d82b10aef6ebafea6e9>:0
  at System.Windows.Forms.Control.CreateControl () [0x00039] in <36fbc38531724d82b10aef6ebafea6e9>:0
  at System.Windows.Forms.Control.SetVisibleCore (bool) [0x0003a] in <36fbc38531724d82b10aef6ebafea6e9>:0
  at System.Windows.Forms.Form.SetVisibleCore (bool) [0x00065] in <36fbc38531724d82b10aef6ebafea6e9>:0
  at System.Windows.Forms.Control.set_Visible (bool) [0x00009] in <36fbc38531724d82b10aef6ebafea6e9>:0
  at System.Windows.Forms.Control.Show () [0x00000] in <36fbc38531724d82b10aef6ebafea6e9>:0
  at (wrapper remoting-invoke-with-check) System.Windows.Forms.Control.Show () [0x00031] in <36fbc38531724d82b10aef6ebafea6e9>:0
  at GerberCombinerBuilder.GerberPanelizerParent..ctor () [0x00017] in <d4a1e563f66b4c33be95409fb7549fff>:0
  at (wrapper remoting-invoke-with-check) GerberCombinerBuilder.GerberPanelizerParent..ctor () [0x00018] in <d4a1e563f66b4c33be95409fb7549fff>:0
  at GerberCombinerBuilder.Program.Main () [0x0000b] in <d4a1e563f66b4c33be95409fb7549fff>:0
  at (wrapper runtime-invoke) object.runtime_invoke_void (object,intptr,intptr,intptr) [0x0004c] in <4fdc5ed61a074cafb49fa42deb20d521>:0

Native stacktrace:

        0   mono                                0x0000000103f4a3d1 mono_handle_native_crash + 257
        1   mono                                0x0000000103fb1e86 altstack_handle_and_restore + 70
        2   SkyLight                            0x00007fff664fd424 SLDisplayBounds + 83
        3   ???                                 0x000000010b2d667e 0x0 + 4482492030
        4   ???                                 0x00000001047df396 0x0 + 4370330518
        5   mono                                0x0000000103ea45f4 mono_jit_runtime_invoke + 1316
        6   mono                                0x000000010409c238 do_runtime_invoke + 88
        7   mono                                0x00000001040982a4 mono_runtime_class_init_full + 996
        8   mono                                0x0000000103e9da45 mono_jit_compile_method_inner + 2485
        9   mono                                0x0000000103ea0903 mono_jit_compile_method_with_opt + 1379
        10  mono                                0x0000000103f4c871 common_call_trampoline + 1217
        11  mono                                0x0000000103f4c379 mono_magic_trampoline + 73
        12  ???                                 0x000000010435739e 0x0 + 4365579166
        13  ???                                 0x000000010b2d5d4b 0x0 + 4482489675

Debug info from gdb:

(lldb) command source -s 0 '/tmp/mono-gdb-commands.BtxnYC'
Executing commands in '/tmp/mono-gdb-commands.BtxnYC'.
(lldb) process attach --pid 78666
warning: (x86_64) /Library/Frameworks/Mono.framework/Versions/5.4.1/lib/mono/4.5/mscorlib.dll.dylib empty dSYM file detected, dSYM was created with an executable with no debug info.
warning: (x86_64) /Library/Frameworks/Mono.framework/Versions/5.4.1/lib/libglib-2.0.0.dylib empty dSYM file detected, dSYM was created with an executable with no debug info.
Process 78666 stopped
* thread #1, name = 'tid_307', queue = 'com.apple.main-thread', stop reason = signal SIGSTOP
    frame #0: 0x00007fff6c46167a libsystem_kernel.dylib`__wait4 + 10
libsystem_kernel.dylib`__wait4:
->  0x7fff6c46167a <+10>: jae    0x7fff6c461684            ; <+20>
    0x7fff6c46167c <+12>: movq   %rax, %rdi
    0x7fff6c46167f <+15>: jmp    0x7fff6c458791            ; cerror
    0x7fff6c461684 <+20>: retq
Target 0: (mono) stopped.

Executable module set to "/Library/Frameworks/Mono.framework/Versions/Current/Commands/mono".
[... cut ...]
bittond:Panelizer/ $ mono -V                                                                                                                                              [11:56:55]
Mono JIT compiler version 5.4.1.7 (2017-06/e66d9abbb27 Wed Oct 25 12:10:41 EDT 2017)
Copyright (C) 2002-2014 Novell, Inc, Xamarin Inc and Contributors. www.mono-project.com
        TLS:           normal
        SIGSEGV:       altstack
        Notification:  kqueue
        Architecture:  amd64
        Disabled:      none
        Misc:          softdebug
        LLVM:          yes(3.6.0svn-mono-master/8b1520c8aae)
        GC:            sgen (concurrent by default)
```

runesoeknudsen, 2017-11-15:
`mono GerberPanelizer.exe` should work, however it seems that there is a general problem with winForms and mac,  see e.g. https://github.com/Homebrew/homebrew-core/issues/14684. 

A way around could be using a virtual machine running Linux. Note that it needs to support OpenGL

davidbitton, 2017-11-15:
If a VM is required, then I'll just run it in my Win10 VM.

macsimski, 2018-01-19:
used runesoeknudsen version under Mono JIT compiler version 5.4.1.7 on Ubuntu 16.04.1 64bit and the generated gerber files contain no mousebites, but the png images have them?? original gerbers generated using kicad

antonio-fiol, 2018-04-28:
Affected by the exception on the description of this bug, I tried @runesoeknudsen 's version on Ubuntu 16.04 with its default version of mono, and it attempts to start (seems like it were going to show a window), but then fails with a bad image format exception. Is it supposed to work with mono 4.2.1.102+dfsg2-7ubuntu4 ?
Currently considering an upgrade to Ubuntu 18.04 soon. This comes with mono 4.6.2.7+dfsg-1ubuntu1 --still behind the version 5.4.0.201 that you tried. Do you think it may help?

CRImier, 2018-12-03:
If someone didn't notice, there are now build instructions that @ulidtko published - #53. Anyone's interested in testing these? Not on Linux rn.

alexws54tk, 2019-03-16:
Nothing works for me
What am I doing wrong?

mono version:
```shell
% mono -V
Mono JIT compiler version 5.18.0.268 (tarball Sat Mar 16 14:25:29 +07 2019)
Copyright (C) 2002-2014 Novell, Inc, Xamarin Inc and Contributors. www.mono-project.com
	TLS:           __thread
	SIGSEGV:       altstack
	Notifications: epoll
	Architecture:  amd64
	Disabled:      none
	Misc:          softdebug 
	Interpreter:   yes
	LLVM:          supported, not enabled.
	Suspend:       preemptive
	GC:            sgen (concurrent by default)
```
error on startup:
```shell
% mono GerberPanelizer.exe 
Unhandled Exception:
System.InvalidCastException: Specified cast is not valid.
  at GerberCombinerBuilder.GerberPanelizerParent.InitializeComponent () [0x002f4] in <19d40ec926044f24bd4f992136fb1d19>:0 
  at GerberCombinerBuilder.GerberPanelizerParent..ctor () [0x00034] in <19d40ec926044f24bd4f992136fb1d19>:0 
  at (wrapper remoting-invoke-with-check) GerberCombinerBuilder.GerberPanelizerParent..ctor()
  at GerberCombinerBuilder.Program.Main () [0x0000e] in <19d40ec926044f24bd4f992136fb1d19>:0 
[ERROR] FATAL UNHANDLED EXCEPTION: System.InvalidCastException: Specified cast is not valid.
  at GerberCombinerBuilder.GerberPanelizerParent.InitializeComponent () [0x002f4] in <19d40ec926044f24bd4f992136fb1d19>:0 
  at GerberCombinerBuilder.GerberPanelizerParent..ctor () [0x00034] in <19d40ec926044f24bd4f992136fb1d19>:0 
  at (wrapper remoting-invoke-with-check) GerberCombinerBuilder.GerberPanelizerParent..ctor()
  at GerberCombinerBuilder.Program.Main () [0x0000e] in <19d40ec926044f24bd4f992136fb1d19>:0 
```

ulidtko, 2019-03-17:
@alexws54tk which source did you build, and how? Can you build a debug version with source line locations

alexws54tk, 2019-03-17:
@ulidtko I build sources from main repo with some patches from #71 trough `build.sh` script file.

> Can you build a debug version with source line locations

```shell
alex@aero [17/03 17:47:09] [~/src/GerberTools] [master *]
-> % xbuild /p:Configuration=Debug GerberProjects/GerberProjects.sln 2>&1 >output
```
[build.log](http://gpo.ws54.tk/gentoo-builds/build.log-GerberTools-2019.9999)

Same error on start.
```shell
alex@aero [17/03 18:32:53] [~/src/GerberTools] [master *]
-> % mono GerberPanelizer/bin/Debug/GerberPanelizer.exe 
Unhandled Exception:
System.InvalidCastException: Specified cast is not valid.
  at GerberCombinerBuilder.GerberPanelizerParent.InitializeComponent () [0x002f4] in <50ae998f472d413888ca5036cc3cce02>:0 
  at GerberCombinerBuilder.GerberPanelizerParent..ctor () [0x00034] in <50ae998f472d413888ca5036cc3cce02>:0 
  at (wrapper remoting-invoke-with-check) GerberCombinerBuilder.GerberPanelizerParent..ctor()
  at GerberCombinerBuilder.Program.Main () [0x0000e] in <50ae998f472d413888ca5036cc3cce02>:0 
[ERROR] FATAL UNHANDLED EXCEPTION: System.InvalidCastException: Specified cast is not valid.
  at GerberCombinerBuilder.GerberPanelizerParent.InitializeComponent () [0x002f4] in <50ae998f472d413888ca5036cc3cce02>:0 
  at GerberCombinerBuilder.GerberPanelizerParent..ctor () [0x00034] in <50ae998f472d413888ca5036cc3cce02>:0 
  at (wrapper remoting-invoke-with-check) GerberCombinerBuilder.GerberPanelizerParent..ctor()
  at GerberCombinerBuilder.Program.Main () [0x0000e] in <50ae998f472d413888ca5036cc3cce02>:0 
```

```shell
alex@aero [17/03 18:43:35] [~/src/GerberTools] [master *]
-> % mono ./ReleaseBuilder.exe DirtyPCBs DirtyPCBs\Build

Directory not found!

alex@aero [17/03 18:43:43] [~/src/GerberTools] [master *]
-> % mono ./ReleaseBuilder.exe GerberTools Build
Unhandled Exception:
System.UriFormatException: Invalid URI: The format of the URI could not be determined.
  at System.Uri.CreateThis (System.String uri, System.Boolean dontEscape, System.UriKind uriKind) [0x0007b] in <9268f3792dc94bbabfe610485b45e6a0>:0 
  at System.Uri..ctor (System.String uriString, System.UriKind uriKind) [0x00014] in <9268f3792dc94bbabfe610485b45e6a0>:0 
  at ReleaseBuilder.ReleaseBuilder.RelativePath (System.String path1, System.String path2) [0x00001] in <ad1089b2f1bb4da088d45d9907849a99>:0 
  at ReleaseBuilder.ReleaseBuilder.AddFilesToFolder (Ionic.Zip.ZipFile zip, System.String inputfolder, System.String basefolder) [0x00046] in <ad1089b2f1bb4da088d45d9907849a99>:0 
  at ReleaseBuilder.ReleaseBuilder.AddFolder (Ionic.Zip.ZipFile zip, System.String inputfolder, System.String basefolder) [0x00035] in <ad1089b2f1bb4da088d45d9907849a99>:0 
  at ReleaseBuilder.ReleaseBuilder.Main (System.String[] args) [0x00114] in <ad1089b2f1bb4da088d45d9907849a99>:0 
[ERROR] FATAL UNHANDLED EXCEPTION: System.UriFormatException: Invalid URI: The format of the URI could not be determined.
  at System.Uri.CreateThis (System.String uri, System.Boolean dontEscape, System.UriKind uriKind) [0x0007b] in <9268f3792dc94bbabfe610485b45e6a0>:0 
  at System.Uri..ctor (System.String uriString, System.UriKind uriKind) [0x00014] in <9268f3792dc94bbabfe610485b45e6a0>:0 
  at ReleaseBuilder.ReleaseBuilder.RelativePath (System.String path1, System.String path2) [0x00001] in <ad1089b2f1bb4da088d45d9907849a99>:0 
  at ReleaseBuilder.ReleaseBuilder.AddFilesToFolder (Ionic.Zip.ZipFile zip, System.String inputfolder, System.String basefolder) [0x00046] in <ad1089b2f1bb4da088d45d9907849a99>:0 
  at ReleaseBuilder.ReleaseBuilder.AddFolder (Ionic.Zip.ZipFile zip, System.String inputfolder, System.String basefolder) [0x00035] in <ad1089b2f1bb4da088d45d9907849a99>:0 
  at ReleaseBuilder.ReleaseBuilder.Main (System.String[] args) [0x00114] in <ad1089b2f1bb4da088d45d9907849a99>:0 
```

ЗЫ: В общем: очередное ненужное подделие с отсутствием заявленной работоспособности. Только зря два дня потерял.

taotieren, 2021-04-26:
Winelib is a development toolkit which allows you to compile your Windows applications on Unix.

https://wiki.winehq.org/Winelib_User's_Guide

Alfystar, 2021-06-04:
## My problem
Hi every one
I'm trying compile te project under linux, in particular ubuntu 20.04.

Unfortunaly i'm getting in truble...
anyway this is my error ouput after the installation of the dotnet and mono-complete packege:

```
/home/alfy/Documents/GitHub-Other/GerberTools/GerberProjects/GerberProjects.sln (default targets) ->
(Build target) ->
/home/alfy/Documents/GitHub-Other/GerberTools/GerberLibrary/GerberLibrary.csproj (default targets) ->
/usr/lib/mono/xbuild/14.0/bin/Microsoft.CSharp.targets (CoreCompile target) ->

        Core/Gerber.cs(14,7): error CS0246: The type or namespace name 'Ionic' could not be found (are you missing a using directive or an assembly reference?)
        Artwork Related/GerberOutlineWriter.cs(4,7): error CS0246: The type or namespace name 'QiHe' could not be found (are you missing a using directive or an assembly reference?)
        Core/GerberPanel.cs(13,7): error CS0246: The type or namespace name 'Ionic' could not be found (are you missing a using directive or an assembly reference?)
        Core/ImageCreator.cs(13,7): error CS0246: The type or namespace name 'Ionic' could not be found (are you missing a using directive or an assembly reference?)
        Core/SickOfBeige.cs(13,7): error CS0246: The type or namespace name 'Ionic' could not be found (are you missing a using directive or an assembly reference?)
```

Any one cane help me??? I realy don't know what i need to install now.

### Update build.sh

I also update the `build.sh` script to automatize the mono packge installation for ubuntu, maybe is possible extend to all the OS.
This is the modify script:
```
#!/bin/bash

set -euo pipefail
set -x

nuget () {
	package="$1"
	version="$2"
	zip="`mktemp`.zip"
	url="https://www.nuget.org/api/v2/package/$package/$version"
	dir="GerberProjects/packages/$package.$version"

	test -d "$dir" || {
		wget "$url" -O "$zip"
		mkdir -p "$dir"
		unzip "$zip" -d "$dir"
		rm "$zip"
	}
}

# Needed as xbuild gets confused about some modern terminals
export TERM=xterm

echo "If 'mono' aren't install in your computer, please follow the most appropriate guide here:"
echo 'https://docs.microsoft.com/it-it/dotnet/core/install/linux'
echo "In particular the 'mono-complete' version, now try install for ubuntu"

UBUNTU_VERSION=$(lsb_release -rs);
REQ_UBUNTU_VERSION_1="20.04";
REQ_UBUNTU_VERSION_1_1="21.04";
REQ_UBUNTU_VERSION_2="18.04";
REQ_UBUNTU_VERSION_3="16.04";

# Mono install
if [[ $UBUNTU_VERSION == $REQ_UBUNTU_VERSION_1 ]] || [[ $UBUNTU_VERSION == $REQ_UBUNTU_VERSION_1_1 ]]; then
	sudo apt install gnupg ca-certificates
	sudo apt-key adv --keyserver hkp://keyserver.ubuntu.com:80 --recv-keys 3FA7E0328081BFF6A14DA29AA6A19B38D3D831EF
	echo "deb https://download.mono-project.com/repo/ubuntu stable-focal main" | sudo tee /etc/apt/sources.list.d/mono-official-stable.list
	sudo apt update
elif [[ $UBUNTU_VERSION == $REQ_UBUNTU_VERSION_2 ]]; then
	sudo apt install gnupg ca-certificates
	sudo apt-key adv --keyserver hkp://keyserver.ubuntu.com:80 --recv-keys 3FA7E0328081BFF6A14DA29AA6A19B38D3D831EF
	echo "deb https://download.mono-project.com/repo/ubuntu stable-bionic main" | sudo tee /etc/apt/sources.list.d/mono-official-stable.list
	sudo apt update
elif [[ $UBUNTU_VERSION == $REQ_UBUNTU_VERSION_3 ]]; then
	sudo apt-key adv --keyserver hkp://keyserver.ubuntu.com:80 --recv-keys 3FA7E0328081BFF6A14DA29AA6A19B38D3D831EF
	sudo apt install apt-transport-https ca-certificates
	echo "deb https://download.mono-project.com/repo/ubuntu stable-xenial main" | sudo tee /etc/apt/sources.list.d/mono-official-stable.list
	sudo apt update
else
	echo "Imppossible select appropriate Mono-version, please download your self"
	exit 1
fi
sudo apt install mono-complete

#dependance packges
nuget Triangle 0.0.6-Beta3
nuget DotNetZip 1.12.0
nuget OpenTK 3.0.1
nuget OpenTK.GLControl 3.0.1
nuget netDXF 2.0.2
nuget netDXF 0.9.3
nuget GlmNet 0.5.1
nuget DockPanelSuite 3.0.6
nuget DockPanelSuite.ThemeVS2015 3.0.6

echo "If 'dotnet' aren't install in your computer, please follow the most appropriate guide here:"
echo 'https://docs.microsoft.com/it-it/dotnet/core/install/linux'

xbuild /p:Configuration=Debug GerberProjects/GerberProjects.sln

```