ThisIsNotRocketScience/GerberTools #15: Crashes on large gerber layers

Lymkin, 2017-03-18:
I cannot panelize my gerbers that have a 23Mb copper layer.  I'm trying to put six on one board and it cannot complete because it crashes.  No indication what is really happening.  Could be memory issue, but I have 16Gb of memory and it isn't using it all.  How about a 64-bit version?

StijnKuipers, 2017-03-18:
Technically it already is 64bit - but .NET seems to have an odd maximum-object-size limit of 2GB regardless. Send me the file and I'll look in to it -> it should not be this fatal.

StijnKuipers, 2017-03-18:
https://msdn.microsoft.com/en-us/library/hh285054(v=vs.110).aspx  This could fix it... will investigate

Lymkin, 2017-03-18:
Here you go.  Gerberset is in the zipfile.  Thanks for looking at this!
[CCB.zip](https://github.com/ThisIsNotRocketScience/GerberTools/files/852486/CCB.zip)

StijnKuipers, 2017-03-19:
Just checked the file -> you are aware that this is a ridiculously large file for what you are trying to achieve? Did you generate this by converting a bitmap to gerber? Looking at the previz, these should have been tiny vector files. While you are at it: the bottom silkscreen looks a bit underdesigned compared to the top side -> consider making the labelling just as nice since you are generating art externally anyway ;-) 

I have reproduced the out-of-memory crash - a quick workaround could be to disable the clipping of silk in the boardproperties - thats the place that needs the most ram to work. New build has 64bit enabled - will upload it soon.

Lymkin, 2017-03-19:
Yep was way to large.  I asked around on the EagleCad forums, and it turns out if you use 0 as the outline width for a polygon, the gerber turns out huge.  I was able to redraw those polygons with a larger width line and they turned out fine.  I knew a simple polygon should not turn out so large.  I hope this is something AutoDesk fixes since they now own EagleCad now.  Perimeter line width should be irrelevant for a polygon.  Thanks for looking at this.  Great program so far!

StijnKuipers, 2017-03-24:
64 bit enabled - but still inadvisable to make tons of 0-width hairlines..

closed 2017-03-24 by StijnKuipers