ThisIsNotRocketScience/GerberTools #21: Panelizer keeps crashing on Windows 10 64bit

TheNitek, 2017-05-14:
Hi there,
I'm trying to get Panelizer working on my machine, but when I open try to open my gerber folder it crashes everytime:

> See the end of this message for details on invoking 
> just-in-time (JIT) debugging instead of this dialog box.
> 
> ************** Exception Text **************
> System.EntryPointNotFoundException: Unable to find an entry point named 'glGenerateMipmap' in DLL 'opengl32.dll'.
>    at OpenTK.Graphics.OpenGL.GL.Core.GenerateMipmap(GenerateMipmapTarget target)
>    at OpenTK.Graphics.OpenGL.GL.GenerateMipmap(GenerateMipmapTarget target)
>    at QuickFont.TexturePage.CreateTexture(BitmapData dataSource)
>    at QuickFont.Builder.BuildFontData(String saveName)
>    at QuickFont.QFont..ctor(Font font, QFontBuilderConfiguration config)
>    at GerberCombinerBuilder.GLGraphicsInterface.CheckFont()
>    at GerberCombinerBuilder.GLGraphicsInterface.MeasureString(String p)
>    at GerberLibrary.GerberPanel.RenderInstance(GraphicsInterface G, Single PW, Color C, AngledThing b, Boolean errors, Boolean active, Boolean hover)
>    at GerberLibrary.GerberPanel.DrawBoardBitmap(Single PW, GraphicsInterface G, Int32 targetwidth, Int32 targetheight, AngledThing SelectedInstance, AngledThing Hoverinstance, Double snapdistance)
>    at GerberCombinerBuilder.GerberPanelize.glControl1_Paint(Object sender, PaintEventArgs e)
>    at System.Windows.Forms.Control.OnPaint(PaintEventArgs e)
>    at OpenTK.GLControl.OnPaint(PaintEventArgs e)
>    at System.Windows.Forms.Control.PaintWithErrorHandling(PaintEventArgs e, Int16 layer)
>    at System.Windows.Forms.Control.WmPaint(Message& m)
>    at System.Windows.Forms.Control.WndProc(Message& m)
>    at System.Windows.Forms.ScrollableControl.WndProc(Message& m)
>    at System.Windows.Forms.ContainerControl.WndProc(Message& m)
>    at System.Windows.Forms.UserControl.WndProc(Message& m)
>    at System.Windows.Forms.Control.ControlNativeWindow.OnMessage(Message& m)
>    at System.Windows.Forms.Control.ControlNativeWindow.WndProc(Message& m)
>    at System.Windows.Forms.NativeWindow.Callback(IntPtr hWnd, Int32 msg, IntPtr wparam, IntPtr lparam)
> 
> 
> ************** Loaded Assemblies **************
> mscorlib
>     Assembly Version: 4.0.0.0
>     Win32 Version: 4.7.2046.0 built by: NET47REL1
>     CodeBase: file:///C:/Windows/Microsoft.NET/Framework/v4.0.30319/mscorlib.dll
> ----------------------------------------
> GerberPanelizer
>     Assembly Version: 1.0.6257.24941
>     Win32 Version: 1.0.*
>     CodeBase: file:///C:/Users/Nitek/Desktop/GerberTools_2017_2_17/Panelizer/GerberPanelizer.exe
> ----------------------------------------
> System.Windows.Forms
>     Assembly Version: 4.0.0.0
>     Win32 Version: 4.7.2094.0 built by: NET47REL1LAST
>     CodeBase: file:///C:/WINDOWS/Microsoft.Net/assembly/GAC_MSIL/System.Windows.Forms/v4.0_4.0.0.0__b77a5c561934e089/System.Windows.Forms.dll
> ----------------------------------------
> System
>     Assembly Version: 4.0.0.0
>     Win32 Version: 4.7.2093.0 built by: NET47REL1LAST
>     CodeBase: file:///C:/WINDOWS/Microsoft.Net/assembly/GAC_MSIL/System/v4.0_4.0.0.0__b77a5c561934e089/System.dll
> ----------------------------------------
> System.Drawing
>     Assembly Version: 4.0.0.0
>     Win32 Version: 4.7.2046.0 built by: NET47REL1
>     CodeBase: file:///C:/WINDOWS/Microsoft.Net/assembly/GAC_MSIL/System.Drawing/v4.0_4.0.0.0__b03f5f7f11d50a3a/System.Drawing.dll
> ----------------------------------------
> System.Configuration
>     Assembly Version: 4.0.0.0
>     Win32 Version: 4.7.2046.0 built by: NET47REL1
>     CodeBase: file:///C:/WINDOWS/Microsoft.Net/assembly/GAC_MSIL/System.Configuration/v4.0_4.0.0.0__b03f5f7f11d50a3a/System.Configuration.dll
> ----------------------------------------
> System.Core
>     Assembly Version: 4.0.0.0
>     Win32 Version: 4.7.2093.0 built by: NET47REL1LAST
>     CodeBase: file:///C:/WINDOWS/Microsoft.Net/assembly/GAC_MSIL/System.Core/v4.0_4.0.0.0__b77a5c561934e089/System.Core.dll
> ----------------------------------------
> System.Xml
>     Assembly Version: 4.0.0.0
>     Win32 Version: 4.7.2046.0 built by: NET47REL1
>     CodeBase: file:///C:/WINDOWS/Microsoft.Net/assembly/GAC_MSIL/System.Xml/v4.0_4.0.0.0__b77a5c561934e089/System.Xml.dll
> ----------------------------------------
> GerberLibrary
>     Assembly Version: 1.0.0.0
>     Win32 Version: 1.0.0.0
>     CodeBase: file:///C:/Users/Nitek/Desktop/GerberTools_2017_2_17/Panelizer/GerberLibrary.DLL
> ----------------------------------------
> OpenTK.GLControl
>     Assembly Version: 1.1.0.0
>     Win32 Version: 1.1.2349.61993
>     CodeBase: file:///C:/Users/Nitek/Desktop/GerberTools_2017_2_17/Panelizer/OpenTK.GLControl.DLL
> ----------------------------------------
> OpenTK
>     Assembly Version: 1.1.0.0
>     Win32 Version: 1.1.2349.61993
>     CodeBase: file:///C:/Users/Nitek/Desktop/GerberTools_2017_2_17/Panelizer/OpenTK.DLL
> ----------------------------------------
> System.DirectoryServices
>     Assembly Version: 4.0.0.0
>     Win32 Version: 4.7.2046.0 built by: NET47REL1
>     CodeBase: file:///C:/WINDOWS/Microsoft.Net/assembly/GAC_MSIL/System.DirectoryServices/v4.0_4.0.0.0__b03f5f7f11d50a3a/System.DirectoryServices.dll
> ----------------------------------------
> Triangle
>     Assembly Version: 0.0.6.0
>     Win32 Version: 0.0.6.0
>     CodeBase: file:///C:/Users/Nitek/Desktop/GerberTools_2017_2_17/Panelizer/Triangle.DLL
> ----------------------------------------
> 
> ************** JIT Debugging **************
> To enable just-in-time (JIT) debugging, the .config file for this
> application or computer (machine.config) must have the
> jitDebugging value set in the system.windows.forms section.
> The application must also be compiled with debugging
> enabled.
> 
> For example:
> 
> <configuration>
>     <system.windows.forms jitDebugging="true" />
> </configuration>
> 
> When JIT debugging is enabled, any unhandled exception
> will be sent to the JIT debugger registered on the computer
> rather than be handled by this dialog box.

StijnKuipers, 2017-05-15:
This is an OpenGL driver issue -> what type of videocard do you have? (and what version of the driver?)

TheNitek, 2017-05-15:
I think so too, but google didn't provide much help :-/
It a Geforce GTX 560 Ti using the Nvidia Driver V382.05

solexious, 2017-05-20:
I have the same issue on Windows 10 64 bit

GTX 970 with driver V382.05

solexious, 2017-05-20:
I find I have this issue on both github releases, including if I try with my windows 10 vm.

But I don't have this issue if I run the pre github version (23-08-2016) and the program works correctly.

deece, 2017-05-23:
Me too, GTX970, Driver 382.05

GumboBeans, 2017-05-25:
Ditto on that one i'm afraid, GTX960 381.89

TheNitek, 2017-06-03:
https://github.com/ThisIsNotRocketScience/GerberTools/commit/67e413644d9d7cbeeadd3ec301e9cfc03dd458d1 Fixed it for me!

HorstBaerbel, 2017-06-22:
I came here by accident. Just a hint, without looking at the code: Automatic mipmap generation has been deprecated on newer (afair >= 4.3) OpenGL versions. I've also seen it crash horribly on older OpenGL implementations / driver versions or older GPUs (see GM965 above), though advertised as "supported" by the OpenGL implementation through the extensions. The bottom line for me was simply: Don't use it.

StijnKuipers, 2017-07-01:
Yeah - I hunted down where it was still used.. - it has been stripped from the source quite a while ago already, but there still needs to be a new release build. Will add a new zip to the site shortly.

oddstr13, 2017-08-30:
Still present in latest release (2017-02-17)
Windows 10 64bit, GTX 1070

patzik, 2017-09-27:
I am also having the same problems, I upgraded my computer windows 10 64bit and get this error.
Also using an nvidia video card 780Ti. Installed a virtual machine but on this the program just crashes...
Any updates or solutions? I really like the program, hope I can use it again soon....

PheiPheiPhei, 2017-10-14:
@StijnKuipers could you please release a new build? I'm still getting the same bug. Thanks!

patzik, 2017-10-24:
For those having the problem with the 'latest' release crashing.
I downloaded the old pre-github version from Stijns website http://blog.thisisnotrocketscience.nl/wp-content/uploads/2015/02/GerberTools-with-framegen.zip and copied
OpenTK.dll
OpenTK.GLControl.dll
to the folder of the 2017-02-17 version and now it doesn't crash anymore.
Hope this helps you too, and still hoping that Stijn will release a new build soon asI tried to compile it myself but failed ;-)

brunobasto, 2017-12-26:
@patzik, that fixed it for me! Thx

patzik, 2017-12-26:
Glad I could help....

bjpirt, 2018-02-02:
I'm getting the same error running on a Windows VirtualBox machine. Complaining about mipmapping as above.

I'm guessing the above fix might help me too but I have no experience of building C# on Windows. If anyone can point me in the direction of a "how to build on Windows guide for noobs" and I'll dig in myself Or alternatively, would there be any chance of a new release?

It looks like great software - very keen to take it for a spin!

patzik, 2018-02-02:
Try the method I decribed with just copying the 2 files. I wasn’t smart enough either to compile it.

bjpirt, 2018-02-02:
I did try that but unfortunately it didn't seem to work for me :-(

merlokk, 2018-04-25:
copying the OpenTK.dll and GLControl.dll works!!!
THANKS!

LowPowerLab, 2018-05-13:
The copy OpenTK.dll & OpenTK.GLControl.dll method worked for me as well.
Otherwise everything crashed. Windows10 GTx 1050ti
Building in VS 2017 works but that build crashes as well.

luftek, 2018-08-06:
Should we donate $ for his time? Donate button on left side: http://blog.thisisnotrocketscience.nl/projects/pcb-panelizer/

I do save $ when using it.

siteswapjuggler, 2020-07-15:
The copy OpenTK.dll & OpenTK.GLControl.dll method worked for me as well.

chreptusio, 2024-04-22:
Keeps crashing on Win7, Win8 and Win11.