ThisIsNotRocketScience/GerberTools #22: Boards with origin point far from 0,0 spawn outside panel area

CRImier, 2017-05-15:
Hi! In KiCad, for example, the drawing area is the size of an A4 sheet, and board element coordinates are calculated from the top left corner. So, a board that's drawn in the center of the KiCad drawing area, will spawn at ~(150, 100), which is well outside the drawing area for most panels (and there might be other factors at play, too, since IIRC I had boards that wouldn't appear without increasing the X/Y numbers even more). It's even worse when rotating boards, because they rotate in relation to the 0,0 point and you have to re-calculate the coordinates. I've described a workaround in this [Gerber Panelizer tutorial](https://hackaday.io/project/19202/log/59210) I did recently, but it would be much more user-friendly if that issue was worked around automatically.

luftek, 2017-05-15:
Why not use auxiliary coordinates and select that option in KiCad? Works great for me.
Use auxiliary axis as origin:
https://cdn.hackaday.io/images/9152151494287348088.PNG

CRImier, 2017-05-15:
Oh. Do I need to also place auxiliary axis reference point somewhere, like, with a tool in PCBNew?

CRImier, 2017-05-15:
https://forum.kicad.info/t/auxiliary-axis-coordinate-origin-preparing-files-for-fabrication-can-someone-explain/1820/4 - like this?

luftek, 2017-05-15:
PcbNew -> Place -> Drill and Place Offset
Naming is kind of weird, but is the same thing = auxiliary axis :)

CRImier, 2017-05-15:
The naming must be the reason why I haven't noticed that option, I tried the "reference point for the grid" thing but it didn't help so I thought KiCad can't deal with it itself. Thank you! Will check it, add to the tutorial and close the issue =)

CRImier, 2017-06-08:
Checked the "auxiliary axis origin" thing in KiCad, works for me, updated my tutorial with it.

closed 2017-06-08 by CRImier