ThisIsNotRocketScience/GerberTools #23: .gm1 file not taken into account

CRImier, 2017-05-15:
KiCad uses .gm1 files instead of .gml (file syntax is identical, copying .gm1 to .gml will work). It's stupid, but IIRC KiCad devs don't want to change that. It would be very useful (and user-friendly) to have a workaround for this. I am thinking about looking into it (should be simple to add), but I'm not sure I'll have the time this or next week.

luftek, 2017-05-15:
I checked for file parser in GerberLibrary\Code\Gerber.cs and found:
on line 353:
              case "gko":
                    Side = BoardSide.Both;
                    Layer = BoardLayer.Outline;
                    break;

Adding gm1 would do the job I think, like:
              case "gm1":
              case "gko":
                    Side = BoardSide.Both;
                    Layer = BoardLayer.Outline;
                    break;

Other *.cs files just use "gko" as export extension.
I don't have the envoironment to build this ready, but I don't find any other extension to collide with this purpose

luftek, 2017-05-15:
or maybe add "gm1" before case "gml":
Don't really know the difference between BoardLayer.Mill and BoardLayer.Outline

TheNitek, 2017-05-15:
According to https://github.com/ThisIsNotRocketScience/GerberTools/issues/9 the gm1 should be gko

CRImier, 2017-05-15:
They might even be the same, for all I know (that's been my experience with gerbers so far, if the tool needs .gko, just rename .gml to .gko)

luftek, 2017-05-15:
Hm, seem like my words of "wisdom". Better ask @StijnKuipers  for differences in Mill and Outline
As said in #9 I renamed .gm1 -> .gko and worked for me with DirtyPCB

luftek, 2017-05-15:
Addeding
**case "gm1":**
would probably solve manual renaming

CRImier, 2017-05-15:
Can you try that, or should I try it myself?

luftek, 2017-05-15:
I won't

StijnKuipers, 2017-05-15:
The outline and mill stage are different during board production

the mill layer is often done -before- electroplating so you can get plated-slots (like for dc-barrel connectors) 

the outline layer is the final cutout of the board. 

So - do the kicad guys have an opinion on which one the gm1/gml is?

CRImier, 2017-05-16:
@StijnKuipers It's hard to understand, the most relevant thing I've found was this [KiCad thread](https://forum.kicad.info/t/oval-pads-not-possible-for-oshpark/835/10). I think the .gm1 file merge will be OK for now - this is the only outline file that KiCad seems to produce for me. If another layer is ever used for mill holes that have to be done before the plating process, I presume that layer is going to be processed automatically (as per #24 ).

deece, 2018-11-28:
Fixed by e7f6fbb7d433cfa8b5459e8587f713d28e371166

CRImier, 2018-12-03:
Closing, then. Thank you!

closed 2018-12-03 by CRImier