ThisIsNotRocketScience/GerberTools #28: Silk outside of outline is printed onto next board

TheNitek, 2017-05-16:
I have a board design which has a silk layer which bigger than the board outline (yes, that is kind of stupid, but KiCad does not really help preventing this). I already ordered single boards an it was no problem. If I use the panelizer with this boards, the silk layer from one board is printed onto the board next to it.
Would be great if the silk stay on the board it belongs to.

StijnKuipers, 2017-05-16:
Check the "clip to outline" option in the board properties - warning though, this is a rather heavy operation (and not what I generally -want- my boards to do -> I keep my art gerbers sepearate from my boards - I need them to bleed over) - which is why it is not enabled by default.

Best practise is to keep your gerber-layers exactly to the board-outline, and use other layers to keep information that should not actually be on the board away from the export..

closed 2017-05-16 by StijnKuipers

TCWORLD, 2017-07-14:
Where is the "clip to outline" option? I can't seem to find it in any of the menus. (This is with the latest release from February).