ThisIsNotRocketScience/GerberTools #29: Gernerate single drill file only

TheNitek, 2017-05-25:
As far as I understood the Panelizer is currently creating 2 drill files (combined.drl and combined.txt). Most services seem to honor only one of those files (I'm right now waiting for a dirtypcb order where I realized to late that, at least in the preview, only the combined.txt holes seem to be made :( ). Would be nice to have an option to have both files merged to prevent any issues like this.

luftek, 2017-05-25:
F**k, already in Production at DirtyPCB and not even ordering for myself

TheNitek, 2017-05-25:
I'm not sure yet that this is really an issue with DirtyPCB, might just be the preview!

luftek, 2017-05-25:
Hopefully, Requested them to make photos before sending. Hopefully they do respond with photos.

TheNitek, 2017-06-03:
Did your panels (or pictures) arrive? Any real issues?

luftek, 2017-06-03:
Everything OK. No problem with production.

TheNitek, 2017-06-09:
I can confirm it worked with dirtypcbs (even though the preview was wrong/incomplete). Still for other services it would be great to have this option.

luftek, 2018-08-06:
I have used panelizer for JLCPCB and no problems with this issue "dual files"
Close the issue?

closed 2020-02-13 by TheNitek