ThisIsNotRocketScience/GerberTools #31: Out of memory on .resx while trying to build on Mac

ubidefeo, 2017-07-02:
hi @StijnKuipers 

Not sure if this is the right place, but since I've been trying to run the tools on Mac with very little success I thought I'd give a shot at compiling the source using VS for Mac.
The reason why I'm not able to run on Mac using Mono is because Homebrew's installation does not include mono32 and I get an error during launch.

In my mind it should be pretty straight forward to compile this using VS, but I get __Out of Memory__ errors on all the `.resx` files, right after the Base64 encoding.

Any chance you have tips on this?
No hurry, let me know if you have time :)

hedefalk, 2017-11-10:
Current homebrew includes 32 bit mono, but 64 is default, just:

`mono --arch=32 /Applications/GerberTools_2017_2_17/Panelizer/GerberPanelizer.exe``

Still doesn't work though :)

https://github.com/ThisIsNotRocketScience/GerberTools/issues/38#issuecomment-343437504

brunobasto, 2017-12-25:
Hey @hedefalk, did you manage to get it working?

hedefalk, 2017-12-25:
@brunobasto Nope, unfortunately not :(

brunobasto, 2017-12-25:
So, you gave up using it or just tried with a windows VM?

Also, just our of curiosity, what would you use it for? Do you produce your own boards at home or is it work related?

hedefalk, 2018-02-12:
@brunobasto I went for scripting instead. I'm doing stuff in KiCad so use this python script for panelizing, http://projects.borg.ch/electronics/kicad/panelize.html

Not entirely happy with it and it's a pain to do mousebites and stuff. I think I'm going to cook up my own scripts with that one as an inspiration. Or try to use it programmatically as a lib - it has a weird input file format that's a bit of a hassle to work with, doing stuff like iterations isn't possible from the input format, but should be possible if using the python tool as an api…

However, I re-visited this project now because I want to panelize a mix of of Kicad and Eagle so want to work with Gerbers this time…

Doing panelization mostly for sending to places like dirtypcbs but also to quicken some stuff when using cnc to drill and chemical etching at home.

I'm going to give this one another shot with Ubuntu on VirtualBox. Let's see…

ubidefeo, 2018-05-03:
I'm back at trying this one, but nothing has changed.
I can't build on Mac and it fails under windows 10 in VMWare, so I may have to go dig an older VM on some backup drive if I want to use it.

I guess I'm gonna have to show up at @StijnKuipers 's with some baked goods and bribe him into making this compile on Mac :D
Anyway it seems that the next version of Cuprum is going to include a paneliser. That would be a breath of fresh air for Mac users