ThisIsNotRocketScience/GerberTools #36: oval through-hole

fastlink30, 2017-08-27:
if  component pin is through-hole with shape oval and drill is oval, output gerber hole is not made!

[image]
[image]
[image]

thatch, 2017-09-05:
I actually just came here to report the same issue in the Panelizer.  Here's a simple repro (look at J1); I don't recall what version I'm using but can check.

[plated_slots_test-fab.zip](https://github.com/ThisIsNotRocketScience/GerberTools/files/1279137/plated_slots_test-fab.zip)

I looked at the resulting gerbers, and there are some odd overlapping drills at 0,0 only on boards that had "oval" holes.

fastlink30, 2017-11-04:
in the last update 1/11 the problem remain, for me this is a big problem, i can't publish on dirtypcb

chaosAD, 2017-11-19:
I was about to raise this issue when I bumped into this one. Any chance this is going to be fixed?

bjpirt, 2018-02-15:
I've just run across this issue when using a build from HEAD - would be great to be able to use oval holes

DIYglenn, 2018-07-11:
Seeing the same now in my issue:
[Missing drill holes](https://github.com/ThisIsNotRocketScience/GerberTools/issues/49)

tfuhub, 2018-09-22:
I think this is a bug that does not handle decimal-point. Try this patch:
```
--- E:\GerberTools-2018-08-10\GerberLibrary\Core\GerberSplitter.cs	Sat Jul 14 15:51:08 2018
+++ E:\GerberTools-2018-08-10_patch\GerberLibrary\Core\GerberSplitter.cs	Thu Sep 20 10:45:04 2018
@@ -263 +263 @@ namespace GerberLibrary
-                    if (char.IsNumber(current) || current == '+' || current == '-')
+                    if (char.IsNumber(current) || current == '+' || current == '-' || (hasdecimalpoint && current == '.'))
```

fastlink30, 2018-09-22:
downloaded & recompiled project (triangle library is not present, i must search the source)

anyway confirmed, with the patch proposed i see the oval holes! :) ... finally..... i can re-start use this tool

thanks!

closed 2018-09-22 by fastlink30

ZericE, 2019-07-06:
I just recompiled (v1.0.7125.30657) and am still having issues with oval holes.  Instead of one oval hole, I'm getting two round holes next to each other.  I'm using kicad, and it's output looks correct from the gerber viewer. Is there some other trick I'm missing?
[image]

andreq, 2019-07-16:
@ZericE dunno if you ever found the solution, but in Kicad when you generate drill file switch from "Use router command" to "User alternate drill mode". 

I have to compile latest code + change that in Kicad to get proper oval holes. Would be nice if the library would support G00 command.

ZericE, 2019-07-16:
@andreq Thanks, I will try changing the drill setting in kicad on my next batch.  For the project I was working on I just had the PCB vendor deal with it.

Shackmeister, 2020-02-05:
Added a PR for G00 and G01 support, so the "User alternate drill mode" is not needed
https://github.com/ThisIsNotRocketScience/GerberTools/pull/102