ThisIsNotRocketScience/GerberTools #38: Support MacOS

sabas1080, 2017-08-30:
Hi

Are there plans to support MacOS?

Thanks

StijnKuipers, 2017-08-31:
Not beyond "it works on mono" - all the commandline tools should already
run fine on mono - the gui stuff I have not tested.

closed 2017-08-31 by sabas1080

hedefalk, 2017-11-10:
The GUI stuff doesn't seem to work on OSX. I found this in my search to panelize gerber files from KiCad, so wanted to try the GerberPanelizer.

I'm total noob at Windows, haven't used either Mono or Wine for ages, but my attempt with Mono is this:

* By default Mono runs with 64-bit and that gives me the following warning and later fail:

```
~ ❯❯❯ mono /Applications/GerberTools_2017_2_17/Panelizer/GerberPanelizer.exe                                                                                                  ⏎
WARNING: The Carbon driver has not been ported to 64bits, and very few parts of Windows.Forms will work properly, or at all
```

and then it churns for a bit and then I get this stacktrace:

```
Unhandled Exception:
System.Reflection.TargetInvocationException: Exception has been thrown by the target of an invocation. ---> System.ArgumentException: XOR data length expected 2046820352, read 1370
Parameter name: stream
  at System.Drawing.Icon.InitFromStreamWithSize (System.IO.Stream stream, System.Int32 width, System.Int32 height) [0x0052e] in <515eb91e24e440fbaf749527276e4141>:0
  at System.Drawing.Icon..ctor (System.Runtime.Serialization.SerializationInfo info, System.Runtime.Serialization.StreamingContext context) [0x00095] in <515eb91e24e440fbaf749527276e4141>:0
  at (wrapper managed-to-native) System.Reflection.MonoCMethod:InternalInvoke (System.Reflection.MonoCMethod,object,object[],System.Exception&)
  at System.Reflection.MonoCMethod.InternalInvoke (System.Object obj, System.Object[] parameters) [0x00002] in <ac812cae460544af83bd6cf54c5eee87>:0
   --- End of inner exception stack trace ---
  at System.Reflection.MonoCMethod.InternalInvoke (System.Object obj, System.Object[] parameters) [0x00014] in <ac812cae460544af83bd6cf54c5eee87>:0
  at System.Reflection.MonoCMethod.DoInvoke (System.Object obj, System.Reflection.BindingFlags invokeAttr, System.Reflection.Binder binder, System.Object[] parameters, System.Globalization.CultureInfo culture) [0x0007a] in <ac812cae460544af83bd6cf54c5eee87>:0
  at System.Reflection.MonoCMethod.Invoke (System.Object obj, System.Reflection.BindingFlags invokeAttr, System.Reflection.Binder binder, System.Object[] parameters, System.Globalization.CultureInfo culture) [0x0002f] in <ac812cae460544af83bd6cf54c5eee87>:0
  at System.Reflection.MethodBase.Invoke (System.Object obj, System.Object[] parameters) [0x00000] in <ac812cae460544af83bd6cf54c5eee87>:0
  at System.Reflection.RuntimeConstructorInfo.SerializationInvoke (System.Object target, System.Runtime.Serialization.SerializationInfo info, System.Runtime.Serialization.StreamingContext context) [0x00000] in <ac812cae460544af83bd6cf54c5eee87>:0
  at System.Runtime.Serialization.ObjectManager.CompleteISerializableObject (System.Object obj, System.Runtime.Serialization.SerializationInfo info, System.Runtime.Serialization.StreamingContext context) [0x00059] in <ac812cae460544af83bd6cf54c5eee87>:0
  at System.Runtime.Serialization.ObjectManager.FixupSpecialObject (System.Runtime.Serialization.ObjectHolder holder) [0x00088] in <ac812cae460544af83bd6cf54c5eee87>:0
  at System.Runtime.Serialization.ObjectManager.DoFixups () [0x0005a] in <ac812cae460544af83bd6cf54c5eee87>:0
  at System.Runtime.Serialization.Formatters.Binary.ObjectReader.Deserialize (System.Runtime.Remoting.Messaging.HeaderHandler handler, System.Runtime.Serialization.Formatters.Binary.__BinaryParser serParser, System.Boolean fCheck, System.Boolean isCrossAppDomain, System.Runtime.Remoting.Messaging.IMethodCallMessage methodCallMessage) [0x00077] in <ac812cae460544af83bd6cf54c5eee87>:0
  at System.Runtime.Serialization.Formatters.Binary.BinaryFormatter.Deserialize (System.IO.Stream serializationStream, System.Runtime.Remoting.Messaging.HeaderHandler handler, System.Boolean fCheck, System.Boolean isCrossAppDomain, System.Runtime.Remoting.Messaging.IMethodCallMessage methodCallMessage) [0x000a2] in <ac812cae460544af83bd6cf54c5eee87>:0
  at System.Runtime.Serialization.Formatters.Binary.BinaryFormatter.Deserialize (System.IO.Stream serializationStream, System.Runtime.Remoting.Messaging.HeaderHandler handler, System.Boolean fCheck, System.Runtime.Remoting.Messaging.IMethodCallMessage methodCallMessage) [0x00000] in <ac812cae460544af83bd6cf54c5eee87>:0
  at System.Runtime.Serialization.Formatters.Binary.BinaryFormatter.Deserialize (System.IO.Stream serializationStream, System.Runtime.Remoting.Messaging.HeaderHandler handler, System.Boolean fCheck) [0x00000] in <ac812cae460544af83bd6cf54c5eee87>:0
  at System.Runtime.Serialization.Formatters.Binary.BinaryFormatter.Deserialize (System.IO.Stream serializationStream, System.Runtime.Remoting.Messaging.HeaderHandler handler) [0x00000] in <ac812cae460544af83bd6cf54c5eee87>:0
  at System.Runtime.Serialization.Formatters.Binary.BinaryFormatter.Deserialize (System.IO.Stream serializationStream) [0x00000] in <ac812cae460544af83bd6cf54c5eee87>:0
  at System.Resources.ResourceReader.DeserializeObject (System.Int32 typeIndex) [0x00019] in <ac812cae460544af83bd6cf54c5eee87>:0
  at System.Resources.ResourceReader._LoadObjectV2 (System.Int32 pos, System.Resources.ResourceTypeCode& typeCode) [0x0035c] in <ac812cae460544af83bd6cf54c5eee87>:0
  at System.Resources.ResourceReader.LoadObjectV2 (System.Int32 pos, System.Resources.ResourceTypeCode& typeCode) [0x00000] in <ac812cae460544af83bd6cf54c5eee87>:0
[... cut ...]
  at System.Resources.ResourceManager.GetObject (System.String name) [0x00000] in <ac812cae460544af83bd6cf54c5eee87>:0
  at GerberCombinerBuilder.GerberPanelizerParent.InitializeComponent () [0x00c08] in <d852c0043225491eb769e518f955ce4e>:0
  at GerberCombinerBuilder.GerberPanelizerParent..ctor () [0x00006] in <d852c0043225491eb769e518f955ce4e>:0
  at (wrapper remoting-invoke-with-check) GerberCombinerBuilder.GerberPanelizerParent:.ctor ()
  at GerberCombinerBuilder.Program.Main () [0x0000b] in <d852c0043225491eb769e518f955ce4e>:0
```

Still seems to be icon related though.

lukecyca, 2020-02-02:
Using the latest Wine Stable 5.0, and installing .NET 4.5 using `winetricks dotnet45`, I'm able to get GerberPanelizer.exe to run.

[image]

However when I open a gerber folder, it crashes with the following:

```
Unhandled Exception: System.AccessViolationException: Attempted to read or write protected memory. This is often an indication that other memory is corrupt.
   at QuickFont.QBitmap.EmptyPixel(BitmapData bitmapData, Int32 px, Int32 py)
   at QuickFont.Builder.<>c.<RetargetGlyphRectangleInwards>b__11_1(BitmapData data, Int32 x, Int32 y)
   at QuickFont.Builder.RetargetGlyphRectangleInwards(BitmapData bitmapData, QFontGlyph glyph, Boolean setYOffset, Byte alphaTolerance)
   at QuickFont.Builder.BuildFontData(String saveName)
   at QuickFont.QFont..ctor(Font font, QFontBuilderConfiguration config)
   at GerberCombinerBuilder.GLGraphicsInterface.CheckFont()
   at GerberCombinerBuilder.GLGraphicsInterface.MeasureString(String p)
   at GerberLibrary.GerberPanel.RenderInstance(GraphicsInterface G, Single PW, Color C, AngledThing b, Boolean errors, Boolean active, Boolean hover)
   at GerberLibrary.GerberPanel.DrawBoardBitmap(Single PW, GraphicsInterface G, Int32 targetwidth, Int32 targetheight, AngledThing SelectedInstance, AngledThing Hoverinstance, Double snapdistance)
   at GerberCombinerBuilder.GerberPanelize.glControl1_Paint(Object sender, PaintEventArgs e)
   at System.Windows.Forms.Control.OnPaint(PaintEventArgs e)
   at OpenTK.GLControl.OnPaint(PaintEventArgs e)
   at System.Windows.Forms.Control.PaintWithErrorHandling(PaintEventArgs e, Int16 layer)
   at System.Windows.Forms.Control.WmPaint(Message& m)
   at System.Windows.Forms.Control.WndProc(Message& m)
   at System.Windows.Forms.ScrollableControl.WndProc(Message& m)
   at System.Windows.Forms.UserControl.WndProc(Message& m)
   at System.Windows.Forms.Control.ControlNativeWindow.OnMessage(Message& m)
   at System.Windows.Forms.Control.ControlNativeWindow.WndProc(Message& m)
   at System.Windows.Forms.NativeWindow.Callback(IntPtr hWnd, Int32 msg, IntPtr wparam, IntPtr lparam)
   at System.Windows.Forms.UnsafeNativeMethods.DispatchMessageW(MSG& msg)
   at System.Windows.Forms.Application.ComponentManager.System.Windows.Forms.UnsafeNativeMethods.IMsoComponentManager.FPushMessageLoop(IntPtr dwComponentID, Int32 reason, Int32 pvLoopData)
   at System.Windows.Forms.Application.ThreadContext.RunMessageLoopInner(Int32 reason, ApplicationContext context)
   at System.Windows.Forms.Application.ThreadContext.RunMessageLoop(Int32 reason, ApplicationContext context)
   at System.Windows.Forms.Application.Run(Form mainForm)
   at GerberCombinerBuilder.Program.Main()

```

dejavu1987, 2023-01-24:
Is anyone able to confirm that "it works on mono" please?

dCSeven, 2025-05-06:
> Is anyone able to confirm that "it works on mono" please?

I tried it with wine 9.
Works with dotnet48; does not work with mono (`winetricks forcemono`). 
May be that I'm too inexperienced using it, but I can't tell.