ThisIsNotRocketScience/GerberTools #40: Feature request: Crop board according to outline

ah01, 2017-10-25:
If I try panelize board with some layers exceed board outline (typically silk screen) it can overflow to next instance. 

_Example single board:_

[image]

_Example panel:_

[image]

It would be nice to have some options to automatically crop/delete exceeding paths.

TheNitek, 2017-10-25:
thats a duplicate: https://github.com/ThisIsNotRocketScience/GerberTools/issues/28

ah01, 2017-10-25:
My bad. I was trying this in last release not in master.

_FYI - Eaxmple with enabled clip to outline:_

[image]

PS: It's working only for silk screen, but i guess that's fine.

TheNitek, 2017-10-25:
Had the same problem. You need to compile the source yourself to get a version which supports it. Let's hope for a new release soon

geocine, 2020-11-01:
I just came across this issue, aside from clipping the `Silk Layer` , I also needed to clip the `Solder Mask` and  `Copper Layers`

geocine, 2020-11-01:
I just came across this issue, aside from clipping the `Silk Layer` , I also needed to clip the `Solder Mask` and  `Copper Layers`

You can modify these lines and compile from source https://github.com/ThisIsNotRocketScience/GerberTools/blob/master/GerberLibrary/Core/GerberPanel.cs#L2402

So in my case I changed it  from 
```csharp
if (Layer == BoardLayer.Silk){
 //...
}
```
to
```csharp
if (Layer == BoardLayer.Silk || Layer == BoardLayer.Copper || Layer == BoardLayer.SolderMask){
 //...
}
```