ThisIsNotRocketScience/GerberTools #42: Board inside another board.

TCWORLD, 2017-11-12:
In one of the designs I am working on, there is a large internal cutout within one of the board (defined on outline layer). 

To save space I want to pack another board inside that cutout. When placing the board and adding break tabs, everything seems fine - I can see the board inside another and the break tabs connect:

[image]

When exporting the merged gerbers, it seems that the internal board ("GPIO Breakout" in the sample image) doesn't make it to the rendered output - it gets deleted:

[image]

Fortunately the board is physically present in the merged gerber files, so the merged gerbers are usable, it's just the rendered picture that is incorrect.

I presume this is a bug in the rendering tool.

StijnKuipers, 2017-11-13:
is that internal cutout redefined on a milling layer by any chance? 

make sure it only exists in the outline layer..

TCWORLD, 2017-11-13:
It's defined only on the outline layer.

Interestingly the empty area fill doesn't go in to that cut-out either, but the break tabs are happy to form.