ThisIsNotRocketScience/GerberTools #46: File error?

DaveCalaway, 2018-02-05:
When i put the Gerber files inside the program, nothing appears. 
What i wrong? 
I used the new version of eagle. I was try the extension .gbr but i received the same result. 
[image]

RandomWon, 2018-06-15:
I think this is because your gerber contains a .GM1 file

marmue, 2018-06-25:
Hi Dave,

not sure if this is a bug, but I noticed that some of my exported gerber work and some not. I used different programs and also same, like eagle. For one export it works for another not.

I found that the working gerber do not have any special aperture like %ADD11R,0.05721X0.02375*%
Maybe this is the problem, as it might not be supported?

Best regards
Marco

luftek, 2018-08-06:
Any progress? Close issue?

closed 2018-08-06 by DaveCalaway

StanislavJochman, 2019-08-04:
Hello I found solution. I described it here #93