ThisIsNotRocketScience/GerberTools #49: Panelizer; Missing drill holes, not supporting NPTH and PTH?

DIYglenn, 2018-07-11:
Hi!
About to order my first batch of panelized PCB's, and thought I'd give this a shot.

I've exported my gerbers from KiCAD, where I got a NPTH.drl and PTH.drl file. 
Is it possible to use these with Panelizer? It seems like it has only used the NPTH file, but as PTH...

DIYglenn, 2018-07-11:
I tried again with combining them as one file, but now the drill holes are all gathered in a corner. Not sure what's doing this. 
[image]

Btw, isn't GKO and GML the same file...?

StijnKuipers, 2018-07-11:
The merger merges files based on extention - drl and drl would become one.
Not sure if it currently looks far 1 main drillfile to merge the mousebites
- but since there is no official standard for plated/nonplated/internal
hole naming.. it was hard to keep these apart automatically.

DIYglenn, 2018-07-11:
Could KiCAD be the issue? I seem to get this design working in panelizer. 
When using two files, the NPTH would be treated as PTH, but at least drilled in the right place.

DIYglenn, 2018-07-11:
I stopped using my export and used the one unedited KiCad project from this GitHub repository:
[AmsToMqttBridge](https://github.com/roarfred/AmsToMqttBridge/tree/master/Electrical/HAN_ESP_TSS721/gerber)

That way I can be sure that my changes to the project isn't interfering. This original project only has one drl file. (There are two holes for a network plug, NPTH/PTH doesn't really matter)

I changed "gm1" to "gml", that's all. 

I then:
- created a new project
- selected the gerber folder
- added another instance
- clicked "max rects" 
- added tabs

Exported, and this is the result:
[image]

[image]

No holes...

DIYglenn, 2018-07-11:
I tried with [this guide.](https://hackaday.io/project/19202-small-pcbs-for-panelizing-tutorial/log/59210-panelization-using-gerberpanelizer-on-windows-linux-possible)

There's a tip for KiCad: 

> About the KiCad origin issue, you might want to export your Gerbers (and drill file) with the "Use auxiliary axis as origin" option. In KiCad you can place that auxiliary origin with one of the buttons in the toolbar.

This did work, I placed a "drill and place offset" at the top left corner, and used auxiliary axis as origin on both. I'm still new to this, so I'm not exactly sure what this means/why this helps.

However it DID help. I think I got all the drill holes except four oval shaped ones for the microUSB port. Anyone that can make sense of that?

Original:
[image]

Exported from Panelizer:
[image]

CRImier, 2018-12-03:
It certainly does support plated and non-plated holes, you just have to export them as .drl and .txt (don't know which is which), there's probably an option to do that in your ECAD software. With KiCad, I could export both kinds of holes IIRC (I certainly could export plated holes), just that without the aux. axis thing, it'd place my board far outside of the GP's 10x10cm area... not that it doesn't do that already, but it was more severe =D Since otherwise you get coordinates relative to the KiCad's work area top left corner (one of these corners, at least),  and the work area is an A4 sheet of paper, so you might have to enter something like -100 and -150 in the "X" and "Y" boxes until you actually see your board on the worksheet.