ThisIsNotRocketScience/GerberTools #52: Breaktab in filled board gap not working as reported on HAD

paulcox, 2018-08-03:
Adding a breaktab doesn't work between a board and a filled board outline as the program doesn't seem to be aware for the filled area. This was reported in the HAD article and apparently a fix was provided but it doesn't seem to be in the latest binary available for download. Perhaps if I built from source I'd find this is fixed but so far haven't found instructions for building, I just see a whole bunch of VS projects and not sure where to start....

paulcox, 2018-08-26:
Here´s an example of what I mean: [image]

paulcox, 2018-08-26:
I can confirm that the workaround proposed by CRImier seems to address this issue.

In GerberLibrary/Core/GerberPanel.cs add the second conditional here

[image]

bueffeljaeger, 2018-11-28:
Does someone have already patched the source to get this feature working and could provide a compiled binary?

CRImier, 2018-12-03:
@bueffeljaeger I compiled this a while ago - http://files.zerophone.org/GerberTools-with-fill-fix.zip , use at your own risk and all that. Also, the fix is merged and there's going to be an official release

CRImier, 2018-12-03:
Also, the exec. version I posted was compiled, what, last year? so it might not have some fixes, i.e. oval cutouts.

deece, 2019-01-02:
Fixed in 395a0b55ae5a7b1a568088ac64853a0590753b34