ThisIsNotRocketScience/GerberTools #54: Kicad 5

DaveCalaway, 2018-08-05:
Hi everyone,
I'm using the kicad 5 version and when i plot the gerber files i see that the all files has the extansions ".GBR".

So using this guide, i see the extensions in kicad: https://goo.gl/PiiejM
Now i saved the gerber files in this "old" configuration, but nothing works....

[image]

luftek, 2018-08-05:
- Replace file ending with .gm1 with .gko
What does not work? I use Kicad v5 and it works as expected

DaveCalaway, 2018-08-05:
>  Replace outline file extension .gm1 with .gko   

Yep, done, but does not work.

This is the error:
[image]

> Do you use auxiliary axis in KiCad? recommend!

I think this is only recommended, right?

luftek, 2018-08-05:
- Operating system?
- what happens if you press Continue?

Save error message in .txt file and append here.
If gerbers are not a secret also append

DaveCalaway, 2018-08-05:
> OS:

 Win 10 Home 64 bit

> GerberViewer:

 i don't know, i tried to load in different web Gerber and JLCPCB and it has not problem.

> what happens if you press Continue?:

Here is the error: https://pastebin.com/0ZY7kxkz
If i press continue the program print a big X like my last screenshot.

luftek, 2018-08-05:
Days ago I used panelizer with JLCPCB service, no problem.
Prepare simple gerber files with your flow that you can share.

DaveCalaway, 2018-08-05:
Same problem with GerberTools...
No problems with JLCPCB.
[gerber.zip](https://github.com/ThisIsNotRocketScience/GerberTools/files/2260640/gerber.zip)

luftek, 2018-08-06:
I tried with your gerbers. No issue. I append a working setup.
[image]

Remark:
- auxiliary coordinates would make thing show up immidiately (current offset is in range of 150mm both axis)
- in KiCad I always select to combine NPTH/PTH holes in one .drl  (as per JLCPCB request/spec)

TEST:
For ease of use 
unpack Panelizer.zip in C:\ folder
run panelizer.exe
-> File/Open -> select C:\Panelizer\gerber\set.gerberset

Photos:

[image]

luftek, 2018-08-06:
Please also check issue #21 for possible solutions.

DaveCalaway, 2018-08-06:
With this solution, https://github.com/ThisIsNotRocketScience/GerberTools/issues/21#issuecomment-339008902 , it's works! I used the original Panelizer not your.
I try to upload the gerber file to JLCPCB, but now it does not show me the preview.
This is the result, is it ok?
[image]
[image]

closed 2018-08-06 by DaveCalaway