ThisIsNotRocketScience/GerberTools #57: Border with fudicals

Misfittech, 2018-08-24:
I wanted to create a panel with some boarders, fudicals and alignment holes. I also want to be able to have the alignment holes drilled in the stencil as an option.

TheReelDill, 2018-11-07:
A bit of a hack, but you can create a border by importing border PCB parts.  For example, create an L shaped frame corner PCB, import x4 instances, rotate and position, then create a frame edge(s) PCB, import and stitch corners to frame edge with T key and add your boards.

trampas, 2018-11-07:
Yes I could....  
What I have done so far is just create the whole panel in PCB CAD tool.  Which has it's own issues but it works.  Mainly it takes hours to setup and draw out the panel.  Then if you change the PCB layout you have to start over, so it is not very attractive option.

CRImier, 2018-12-03:
I think the largest problem with this is generating the .pos file for pick-n-placing afterwards, you need to basically do the same with .csv's that panelizer does with gerbers, but account for offsets. Would be cool to add support for concatenating .csv files - i.e. KiCad .csv format for pick&placing.

Misfittech, 2018-12-03:
Well my Pick and Place machine allows the offsets to be done at the machine.  That is if you put fudicals on each board you can align to those for each board in the panel.  Otherwise you can pick components to center on.   Using this I generate PnP files for one board. Then configure the panel at the PnP machine. This also allows me to selectively populate boards in the matrix.  For example one of the boards in the panel may not have passed electrical test, so I disable populating that board on the panel.  

I have not know anyone that generates PnP position files for the panel as whole.

CRImier, 2018-12-03:
The point of fiducials on edges is to avoid having them on the boards, pretty much. We have a similar setup with Liteplacer which allows to do everything you've mentioned, but - it does not allow for rotation, and if you panelize different PCBs (i.e. restocking your Tindie store in one go), it's even harder. Were GerberTools able to process PCBs (or maybe we could have a helper tool that parses GerberTools file), that would not be a problem. I could probably write a helper tool in Python, actually =)

Misfittech, 2018-12-03:
So normally my process is to align the panel with fiducials on the panel wings/border.  Then in my machine I have to specify the offset and rotation for each board in the panel. This is done by picking one or two points on each board in the panel (two for rotation & translation, one for just translation). These two points can be fiducials on the boards themself, or two components/vias on the board. I often use capacitors/resistors but fiducials offer better centering and alignment. 

I have just started creating panels in Altium, which is pain but it works. The reason for this is that:
1. I can add alignment holes where I want and have the stencil printed with alignment holes. 
2. I can make sure I have spacing for components like USB connectors or thumb switches. 
3. I can order PCBs from any vendor and reuse same stencil. 

I also have found that if I take a MDF board and drill holes of the panel alignment and then put cut off finishing nails as pins in holes then I can lay an unframed stencil over board using alignment holes and get perfect stencil alignments instantly.  

I also find that for my stuff I just panelize 4 boards, my parts count are high per board so if did any larger matrix then I waste a lot of parts if I get something wrong.  Hence I find the 2x2 matrix suits my needs at the moment.  I did build a 3x3 panel once and cursed, as that setting up PnP always seems to take one panel of parts to get right and that is a huge amount of waste when you just need to build ~25 boards.

Again I wish there was an easier way to do the panels but I have not found one.  I often wonder what board houses use.  I really hate Altium as that when you update your PCB design you are starting over on the panel...

CRImier, 2018-12-03:
What's your PnP? Liteplacer, OpenPnP build or something else? From workflow description, doesn't sound like Liteplacer. I believe that, if we get tools for making panels, we also need to have tools for generating .csv and other files from those same panels. Your Altium workflow sounds like a pain, indeed.
Don't the stencils generally get generated from soldermask layers, anyway? As you can just send gerbers and receive stencils. My experience is with cheap CN fabs, though.
I'm guessing the fabs have their own custom and/or proprietary software to do panels, they have it for all the other tasks anyway.

StijnKuipers, 2018-12-03:
I have recently started on BOM-merging tools + part renaming schemes... since if you have 10x R1 on a panel, but they might belong to different source boards - this would collide.. 

But if the parts are actually labelled on the board - what should be done with those?

CRImier, 2018-12-04:
That's actually really simple. Now, I'm not speaking about BOM, I'm speaking about POS files for pick&placing- they're the ones where position actually matters (just in case there's a misunderstanding, since you mention BOM). What Liteplacer software does with it's (limited) panelization feature is pretty much "R1-A", and I'm sure that's what you should do too,. Don't stress about it, just append numbers or letters after a dash. If you can't decypher which letter is for which board, in P&P (both LP and OpenPnP can do this) you can navigate to the component with this letter and see which board the nozzle is above of. If you just send the POS to the fab (and use this panelizer for some reason), it's going to work too - they'll jsut load parts into trays and make the machine place them to coordinates, no big deal. As for labels, don't mind the labels - they're not for pick&placing either way, they're for humans.

StijnKuipers, 2018-12-04:
I can add several renaming/numbering schemes - adding A/B/C for various instances should do the trick - or maybe a more descriptive name even

The POS files are very easy to merge - but I'd need a set of commonly accepted variants.. Only weird & ancient Ultiboard is now in there because we had direct need for that..

CRImier, 2018-12-04:
...KiCad? =D
[back_pcb.zip](https://github.com/ThisIsNotRocketScience/GerberTools/files/2641814/back_pcb.zip)
[sh1106_1.3_spi_breakout.zip](https://github.com/ThisIsNotRocketScience/GerberTools/files/2641827/sh1106_1.3_spi_breakout.zip)

Adding two boards with gerbers (should be up-to-date) and .pos generated - back_pcb has one merged file (-all.pos_ , sh1106 has two, one for top and one for bottom (but you can merge those two (top+bottom) safely, in fact, I made them separate in case you want to add such merging). sh1106 might have slightly different origins (changed between gerber and .csv generation), lmk if that's indeed the case, I'll re-send them. I'm really curious what you'll do if you get .pos files from two different EDAs =D

StijnKuipers, 2018-12-04:
Oh and as an aside: the gerberset file is just some xml - should be easy to parse.

CRImier, 2018-12-04:
Yeah, already had to edit it a bunch of times when things broke, mostly because of missing files and whatnot =) Offtopic-ish question - is the XML's format set in stone? What do you do when you're adding options to it, how do you provide backwards compatibility?

Misfittech, 2018-12-04:
I have a Manncorp PnP I got at a 'fire sale' when company went out of business. Before that I built my own PnP and tried OpenPnP, but ended up writing my own PnP software in Python.  In the Python version I did affine transformations for the PCB, that it is it translation, scaling and rotation.   

However the home built PnP was not good enough for what I needed. The Manncorp however rocks, sure it is 12 years old but it places 0402 with no problems.

CRImier, 2019-01-30:
Wrote a script that takes a .gerberset file and merges the .pos files from KiCad into a single .pos file: https://github.com/CRImier/Panelizer2PnP/ Allows me to assemble a panel like this:

[image]

The separate fiducial boards were also done in Kicad and also have a .pos file - one could also use dedicated fiducials on the boards in panel, I just didn't add fiducials to any of the boards on the panel and added the fiducial boards instead.

trampas, 2019-01-30:
I would recommend you add a square border around the panel, especially if
you use a Pick and Place machine as it will allow something to clamp to,
also having holes in boarder for alignment with stencil printer.

CRImier, 2019-01-30:
Thank you for the suggestion! I don't yet use a stencil printer (I use custom lasercut frames), will think about the frame - I have my own contraptions for pick&placing (lasercut jigs, magnets and stuff), but will think about adding frames, or at least using the "fill" option of GerberPaneliser (at least, as soon as the fill tab fix is released as an executable, or I get around to compiling it).