ThisIsNotRocketScience/GerberTools #58: can't import Gerbers

thijsdeschildre, 2018-08-26:
So you just drag and drop a directory with gerbers?
This doesn't work for me. In the upper-right tree I can see the files I add, but I don't see anything shape or part on the grid. I tried on 2 different PC's, both files from Eagle and Kicad, with either GerberTools_2018_8_10 and GerberTools_2017_2_17.
Can someone share a set of Gerber files, so I can see if my exports are to blame?

belenhed, 2018-08-30:
I'm having the same issue :(

Gerber file (RS-274X) generated with Frtizing Version 0.9.3

StijnKuipers, 2018-08-30:
there is a folder "help" inside the gerberpanelizer that has some comments on how to use this.

thijsdeschildre, 2018-08-30:
No, I don't get an outline or anything.

StijnKuipers, 2018-08-30:
the question is -> does your gerber set contain a gko layer?

the doc describes the expected layers - there is only so much magic that can be done automatically on gerber files...

thijsdeschildre, 2018-08-30:
Ah, no, it doesn't. Amazingly, Eagle now has a menu option "generate CAM data" which is lauded as "one-click  manufacturing export".  This worked for me, with PCBway and OSH park, if I recall correctly.
I manually exported layer 20 (outline) in the CAM processor to outline.gko, and now it seems to work. I hope that explains, thanks for helping!

belenhed, 2018-08-31:
mine doesn't either I need to know how to do it in Fritzing

thijsdeschildre, 2018-08-31:
Something is still wrong, and I have no clue what. 
The outlines get processed, but doesn't include the layers, vias, holes,...
Posting screenshots of before/during/after panelizing, with files in each directory.

[image]

[image]

[image]

[image]

LeviSchuck, 2018-09-10:
With KiCad, even with the portal extensions, it uses gm1 instead of gko. 
Had to do lots of renames in my various PCBs to get this to work. 
Although I gave up and just ordered all the tiny pcbs separately because of other problems. 

KiCad used suffixes like `-B.Cu.gbr` for the various layers by default. That could be an enhancement here.

deece, 2019-01-02:
gm1 is now added as an alias of gko. If you can share you gerbers, I can test with the dev version & potentially contribute more fixes if it still does not process correctly.

thijsdeschildre, 2019-03-31:
I tried again. Can't get it to work with either Eagle or Kicad.
Can someone share a project and it's Gerber files that are known to work? That way I can compare with my own files.

30penny, 2020-11-23:
Using Kicad and the plot utility to get drill and gbr files, I then rename everything per the help file and I can see something has loaded in directory view and I see the board dimensions show up, after dropping the one folder with just the one projects gerbers and drill files into the workspace I get no image. I have exported an image using "export merged gerbers" and it does show a png file of my board all layers and drills but I cannot see it in the main workspace to add panelizing or mouse bites.  Any thoughts ?