ThisIsNotRocketScience/GerberTools #70: Feature request - flipping the board (top=>bottom, bottom=>top)

CRImier, 2018-12-03:
Sometimes, a board with weird shape just won't fit inside another board, but it would if it were flipped. I imagine it would be simple to implement, given that you already have a `.RotateFlip` method in code. If I infer its purpose correctly, it should be easy to apply `.Rotate` and `RotateFlip` conditionally based on a checkbox state.

StijnKuipers, 2018-12-04:
The RotateFlip you see is for the standard .NET Bitmap class... 

Flipping a gerber is a bit more involved (to remain compatible with the largest set of readers, macros would need to be flipped as well as polygons reversed...)

CRImier, 2018-12-04:
Oh no. Well, there go my hopes ;-P Thank you for the clarification!