ThisIsNotRocketScience/GerberTools #71: Linux build failed

kprasadvnsi, 2018-12-08:
cloned the repo and run build.sh the build failed. any suggestions ?
[image]

kprasadvnsi, 2018-12-13:
I don't use windows, don't know C# also know nothing about Mono. There is just too many errors to process.

elcojacobs, 2018-12-13:
Thanks for responding :) I think the error from vec2 is solved by adding `using vec2 = Vector<2>`, some errors are resolved by changing \ to / in the project files, but still others remain. I think due to version mismatch between project files (v15) and current msbuild (v16).

I don't have time to further look into it either.

deece, 2019-01-02:
(I added the build script) I didn't see this in my build environment (Fedora 29), however, even if you get past this, there are other issues that need to be addressed, namely weird icon files embedded in the resources that confuse Mono. These need to be extracted, processed with Gimp & re-added to the resources:
https://github.com/ThisIsNotRocketScience/GerberTools/pull/41#issuecomment-442590322

kprasadvnsi, 2019-01-02:
@deece  I don't know C# and don't want to ever use windows for anything. If anybody could document the project structure then i can re-implement the whole thing in different language which provide better compatibility across different platforms. 

I know it's too much to ask but i see no harm in asking.

deece, 2019-01-02:
@kprasadvnsi The resource files are XML containers. You can extract the embedded icons with a text editor, base64 decode them, load them in Gimp, resave them. base64 encode them and paste it back into the resource file. You don't need Windows for that.

keeferty, 2019-01-18:
Did some diagnosis, and it looks like at this point it will not run on linux/osx.
The RESX files are easily fixed by running them through this converter http://converter.webtranslateit.com/.
However since DockPannel seems to be windows only (https://sourceforge.net/p/dockpanelsuite/discussion/402316/thread/ea083210/) i dont see a way of making it work in linux.

deece, 2019-01-19:
@keeferty Thanks for doing the research, would you be able to submit a PR for the resource files? It brings us one step closer to a working Linux build (even if we can't run it), which at the very least could enable some CI.

ole00, 2019-03-16:
There are few issues with the build:
1) resource files: *.resx - they contain embedded images (usus=ally icons) that are specified as "bytearray", but fail to compile . Changing the type to "binary" fixes the compilation. These are:

	modified:   FitBitmapToOutlineAndMerge/FitBitmapToOutlineAndMergeForm.resx
	modified:   GerberPanelizer/Explanation And Warning.resx
	modified:   GerberPanelizer/GerberPanelizerParent.resx
	modified:   GerberPanelizer/PanelProperties.resx
	modified:   GerberPanelizer/Progress.resx
	modified:   GerberPanelizer/WelcomeForm.resx
	modified:   GerberViewer/GerberViewerMainForm.resx
	modified:   GerberViewer/LayerDisplay.resx
	modified:   GerberViewer/LayerList.resx
	modified:   Project_Utilities/IconBuilder/IconFrame.resx
	modified:   Project_Utilities/IconBuilder/IconHost.resx
	modified:   QuickGerberRender/Progress.resx
	modified:   QuickGerberRender/QuickGerberRender.resx

2) incorrect versioning of libraries in GerberViewer/GerberViewer.csproj
```
-    <Reference Include="WeifenLuo.WinFormsUI.Docking, Version=2.11.0.0, Culture=neutral, PublicKeyToken=5cded1a1a0a7b481, processorArchitecture=MSIL">
-      <HintPath>..\GerberProjects\packages\DockPanelSuite.2.11.0\lib\net40\WeifenLuo.WinFormsUI.Docking.dll</HintPath>
+    <Reference Include="WeifenLuo.WinFormsUI.Docking, Version=3.0.6.0, Culture=neutral, PublicKeyToken=5cded1a1a0a7b481, processorArchitecture=MSIL">
+      <HintPath>..\GerberProjects\packages\DockPanelSuite.3.0.6\lib\net40\WeifenLuo.WinFormsUI.Docking.dll</HintPath>
       <Private>True</Private>
     </Reference>
-    <Reference Include="WeifenLuo.WinFormsUI.Docking.ThemeVS2015, Version=2.11.0.0, Culture=neutral, PublicKeyToken=5cded1a1a0a7b481, processorArchitecture=MSIL">
-      <HintPath>..\GerberProjects\packages\DockPanelSuite.ThemeVS2015.2.11.0\lib\net40\WeifenLuo.WinFormsUI.Docking.ThemeVS2015.dll</HintPath>
+    <Reference Include="WeifenLuo.WinFormsUI.Docking.ThemeVS2015, Version=3.0.6.0, Culture=neutral, PublicKeyToken=5cded1a1a0a7b481, processorArchitecture=MSIL">
+      <HintPath>..\GerberProjects\packages\DockPanelSuite.ThemeVS2015.3.0.6\lib\net40\WeifenLuo.WinFormsUI.Docking.ThemeVS2015.dll</HintPath>
       <Private>True</Private>
```

3) Typo in: Project_Utilities/TilingLibrary/TINRS-ArtWork.csproj
```
-      <HintPath>..\..\GerberProjects\packages\GlmNet.0.5.1.0\lib\net40\GlmNet.dll</HintPath>
+      <HintPath>..\..\GerberProjects\packages\GlmNet.0.5.1\lib\net40\GlmNet.dll</HintPath>
```
4) build.sh
```
-#!/bin/sh
+#!/bin/bash
```

These changes fixed the compilation for me (xubuntu 18.10). The executables can be found in <tool>/bin/Debug/ directory.

deece, 2019-03-16:
You shouldn't need the changes to build.sh - what problem were you trying to solve with this?

The other changes look good, please send a PR

ole00, 2019-03-16:
If build.sh is left with /bin/sh I get this error:
<pre>
./build.sh: 3: set: Illegal option -o pipefail
</pre>
and the script does nothing - just exists without attempting to build at all

with /bin/bash it works fine on my linux machine.

alexws54tk, 2019-03-16:
@ole00 
> resource files: *.resx - they contain embedded images (ususally icons) that are specified as "bytearray", but fail to compile . Changing the type to "binary" fixes the compilation. These are: [...]

What kind of changes do you use?

deece, 2019-03-16:
Thanks, since there's no pipes in the scripts, we can remove -o pipefail from the script without losing anything.

We can also remove set -x, as I used that while writing the script and forgot to remove it when i committed it.

ole00, 2019-03-19:
> @ole00
>
>
> What kind of changes do you use?
All of them are the same, like that:
-  <data name="$this.Icon" type="System.Drawing.Icon, System.Drawing" mimetype="application/x-microsoft.net.object.bytearray.base64">
+  <data name="$this.Icon" type="System.Drawing.Icon, System.Drawing" mimetype="application/x-microsoft.net.object.binary.base64">

As I wrote earlier, change the .bytearray. to .binary.

positron96, 2021-01-13:
Hi all, I've managed to build the project in ubuntu. Stuffed the changes into PR: https://github.com/ThisIsNotRocketScience/GerberTools/pull/128/files

Dunno if the patch breaks compilation on windows, hopefully not.

positron96, 2021-01-14:
@ole00 I've done the `bytearray` replacement in the patch above, but noticed that with this fix:
* the program compiles without errors,
* icons are not shown in actual app, whether the exe is run on windows or in linux.
* I can actually change `bytearray` to anything, e.g. `bbb`, the program would still compile fine, but the icons would not be shown.

taotieren, 2021-04-26:
Winelib is a development toolkit which allows you to compile your Windows applications on Unix.

https://wiki.winehq.org/Winelib_User's_Guide