ThisIsNotRocketScience/GerberTools #72: NC drill file (.txt) data is lost when merging files.

kiva113, 2018-12-18:
I'm trying to generate a panel using the gerbers I got from altium. 

The rendered image on export is missing all the drill holes (except for the breakaways) and this is confirmed also with the NC drill file that is missing all the previous information/locations.

I have simplified it to the point where I just add a single board and then have it export. I've tried the august release and the previous older build releases.

Is there a setting or issue with using gerber exports from Altium for use in panelizer?

deece, 2019-01-02:
Can you share your gerbers (or reproduce it with a trivial gerber you can share)? I can test it with the dev build and possibly contribute a fix.

kiva113, 2019-01-20:
Here is a simple board that should have everything.
[https://drive.google.com/file/d/1WwuHvRtqAlxcFoaLv49vMJLH9BHCnY_E/view?usp=sharing](testboard gerbers)

kuya123, 2019-03-30:
I also lost drill file when exporting the merged gerber file.   I was using Altium designer 14

deece, 2019-03-31:
The latest development version should recognise .txt as drill files. Please try building from the git repo, it should fix the issue.