ThisIsNotRocketScience/GerberTools #73: Time for a new release?

deece, 2019-01-02:
@StijnKuipers With all the fixes that have been merged, a new release would be useful :)

I think I fixed the bad package hint paths that required this in your environment: 7af196ab98ed27396be88d2f42999f20badfcc75

TheReelDill, 2019-03-25:
+1 for that.

andreq, 2019-07-16:
+1, would be nice not to have to compile the whole thing to get Panelizer kicad slot fixes :)

mplacona, 2020-07-29:
I am trying to compile this but can't! Can anyone please send me their compiled version? Thanks!

deece, 2020-07-29:
Latest release is here: https://github.com/ThisIsNotRocketScience/GerberTools/releases/tag/20-09-2019

mplacona, 2020-07-30:
Yeah, I’ve seen that. But with it being from September, it doesn’t have the fix you did in November which seems to address the problem I am having with fill empty area.

Would appreciate if you could send me a new release. 

Thanks

deece, 2020-07-30:
Ah, @StijnKuipers will have to help with that.

StijnKuipers, 2020-07-30:
Uploading.

StijnKuipers, 2020-07-30:
https://github.com/ThisIsNotRocketScience/GerberTools/releases/tag/July2020

closed 2020-07-30 by StijnKuipers

mplacona, 2020-07-30:
Totally solves the problem.

Thank you so much!