ThisIsNotRocketScience/GerberTools #84: Linux build failed - debian/ubuntu

nardev, 2019-04-16:
I just tried to build GerberTools but first obstacle was the "#!/bin/sh" in build.sh. I changed that to "#!/bin/bash" and it started that build. However, now i have another issue:

Here is what i get 
[gerbertools.install.log](https://github.com/ThisIsNotRocketScience/GerberTools/files/3083162/gerbertools.install.log)

fiskusmati, 2019-04-18:
I'm running Ubuntu 18.04 and get problems with building:
<pre><font color="#8AE234"><b>user1@user1-Vostro-3550</b></font>:<font color="#729FCF"><b>~/Downloads/GerberTools-master</b></font>$ sudo ./build.sh
./build.sh: 3: set: Illegal option -o pipefail
</pre>

mmoreaux, 2019-05-04:
@fiskusmati `apt install mono-xbuild` will clear _that_ error up.

Meanwhile, on Debian testing (to be Debian 10), I'm running into these issues:

	Core/GerberPanel.cs(13,7): error CS0246: The type or namespace name `Ionic' could not be found. Are you missing an assembly reference?
	Core/GerberPanel.cs(14,7): error CS0246: The type or namespace name `TriangleNet' could not be found. Are you missing an assembly reference?
	Core/GerberPanel.cs(15,7): error CS0246: The type or namespace name `TriangleNet' could not be found. Are you missing an assembly reference?
	Core/ImageCreator.cs(13,7): error CS0246: The type or namespace name `Ionic' could not be found. Are you missing an assembly reference?
	Core/ParsedGerber.cs(8,7): error CS0246: The type or namespace name `TriangleNet' could not be found. Are you missing an assembly reference?
	Core/ParsedGerber.cs(13,7): error CS0246: The type or namespace name `TriangleNet' could not be found. Are you missing an assembly reference?
	Core/ParsedGerber.cs(14,7): error CS0246: The type or namespace name `TriangleNet' could not be found. Are you missing an assembly reference?
	Core/ParsedGerber.cs(15,7): error CS0246: The type or namespace name `TriangleNet' could not be found. Are you missing an assembly reference?
	Core/SickOfBeige.cs(13,7): error CS0246: The type or namespace name `Ionic' could not be found. Are you missing an assembly reference?

I've been trying to replace the `nuget` function in `build.sh` with the actual Nuget package manager to get decent dependency management, but I'm stumbling from one problem into the next...

EuphoricCatface, 2019-05-05:
I'm getting same errors as @nardev. 

```
...

Build FAILED.

Warnings:

/home/[username]/kicad/GerberTools/GerberProjects/GerberProjects.sln:  warning : Don't know how to handle GlobalSection Performance, Ignoring.

/home/[username]/kicad/GerberTools/GerberProjects/GerberProjects.sln (default targets) ->
(Build target) ->
/home/[username]/kicad/GerberTools/GerberLibrary/GerberLibrary.csproj (default targets) ->
/usr/lib/mono/xbuild/14.0/bin/Microsoft.Common.targets (ResolveAssemblyReferences target) ->

	/usr/lib/mono/xbuild/14.0/bin/Microsoft.Common.targets:  warning : Reference 'DotNetZip, Version=1.12.0.0, Culture=neutral, PublicKeyToken=6583c7c814667745, processorArchitecture=MSIL' not resolved
	/usr/lib/mono/xbuild/14.0/bin/Microsoft.Common.targets:  warning : Reference 'System.IO.Compression' not resolved
	/usr/lib/mono/xbuild/14.0/bin/Microsoft.Common.targets:  warning : Reference 'System.IO.Compression.FileSystem' not resolved
	/usr/lib/mono/xbuild/14.0/bin/Microsoft.Common.targets:  warning : Reference 'System.Data.DataSetExtensions' not resolved
	/usr/lib/mono/xbuild/14.0/bin/Microsoft.Common.targets:  warning : Reference 'Triangle, Version=0.0.6.0, Culture=neutral, processorArchitecture=MSIL' not resolved

/home/[username]/kicad/GerberTools/GerberProjects/GerberProjects.sln (default targets) ->
(Build target) ->
/home/[username]/kicad/GerberTools/DirtyPCBs/Base64Extractor/DirtyPCB_Base64Extractor.csproj (default targets) ->
/usr/lib/mono/xbuild/14.0/bin/Microsoft.Common.targets (ResolveAssemblyReferences target) ->

	/usr/lib/mono/xbuild/14.0/bin/Microsoft.Common.targets:  warning : Reference 'System.Data.DataSetExtensions' not resolved
	/usr/lib/mono/xbuild/14.0/bin/Microsoft.Common.targets:  warning : Reference 'System.Net.Http' not resolved

	 8 Warning(s)
	 0 Error(s)

Time Elapsed 00:00:03.0446890
```
Running under Crostini on ChromeOS ARM variant, though I don't believe that's causing the problem.
I know not much about mono, but tried `nuget restore` on the .sln file. All the installation of each package seems to go well, but it still makes same "not resolved" error.

EuphoricCatface, 2019-05-05:
installed `mono-complete` than just `mono-xbuild`. It seems to go a couple of steps further, but that's mostly about it.
```
...

Build FAILED.

Warnings:

/home/[username]/kicad/GerberTools/GerberProjects/GerberProjects.sln:  warning : Don't know how to handle GlobalSection Performance, Ignoring.

/home/[username]/kicad/GerberTools/GerberProjects/GerberProjects.sln (default targets) ->
(Build target) ->
/home/[username]/kicad/GerberTools/GerberLibrary/GerberLibrary.csproj (default targets) ->
/usr/lib/mono/xbuild/14.0/bin/Microsoft.Common.targets (ResolveAssemblyReferences target) ->

	/usr/lib/mono/xbuild/14.0/bin/Microsoft.Common.targets:  warning : Reference 'DotNetZip, Version=1.12.0.0, Culture=neutral, PublicKeyToken=6583c7c814667745, processorArchitecture=MSIL' not resolved
	/usr/lib/mono/xbuild/14.0/bin/Microsoft.Common.targets:  warning : Reference 'Triangle, Version=0.0.6.0, Culture=neutral, processorArchitecture=MSIL' not resolved

/home/[username]/kicad/GerberTools/GerberProjects/GerberProjects.sln (default targets) ->
(Build target) ->
/home/[username]/kicad/GerberTools/GerberLibrary/GerberLibrary.csproj (default targets) ->
/usr/lib/mono/xbuild/14.0/bin/Microsoft.CSharp.targets (CoreCompile target) ->

	Core/Primitives/PointD.cs(10,18): warning CS0661: `GerberLibrary.Core.Primitives.PointD' defines operator == or operator != but does not override Object.GetHashCode()
	Core/Primitives/PointD.cs(10,18): warning CS0659: `GerberLibrary.Core.Primitives.PointD' overrides Object.Equals(object) but does not override Object.GetHashCode()

Errors:

/home/[username]/kicad/GerberTools/GerberProjects/GerberProjects.sln (default targets) ->
(Build target) ->
/home/[username]/kicad/GerberTools/GerberLibrary/GerberLibrary.csproj (default targets) ->
/usr/lib/mono/xbuild/14.0/bin/Microsoft.CSharp.targets (CoreCompile target) ->

	Core/GerberPanel.cs(13,7): error CS0246: The type or namespace name `Ionic' could not be found. Are you missing an assembly reference?
	Core/GerberPanel.cs(14,7): error CS0246: The type or namespace name `TriangleNet' could not be found. Are you missing an assembly reference?
	Core/GerberPanel.cs(15,7): error CS0246: The type or namespace name `TriangleNet' could not be found. Are you missing an assembly reference?
	Core/ImageCreator.cs(13,7): error CS0246: The type or namespace name `Ionic' could not be found. Are you missing an assembly reference?
	Core/ParsedGerber.cs(8,7): error CS0246: The type or namespace name `TriangleNet' could not be found. Are you missing an assembly reference?
	Core/ParsedGerber.cs(13,7): error CS0246: The type or namespace name `TriangleNet' could not be found. Are you missing an assembly reference?
	Core/ParsedGerber.cs(14,7): error CS0246: The type or namespace name `TriangleNet' could not be found. Are you missing an assembly reference?
	Core/ParsedGerber.cs(15,7): error CS0246: The type or namespace name `TriangleNet' could not be found. Are you missing an assembly reference?
	Core/SickOfBeige.cs(13,7): error CS0246: The type or namespace name `Ionic' could not be found. Are you missing an assembly reference?

	 5 Warning(s)
	 9 Error(s)

Time Elapsed 00:00:04.6827050
```

CGrassin, 2019-05-05:
I have the same build issue under Linux Mint with `mono-complete` installed.

EuphoricCatface, 2019-05-12:
Downloaded 2018-08-10 source. fails in similar fashion except DotNetZip version shows 1.11.0.0
Reverted local git to "Build and dependency fixes." 7af196ab98ed27396be88d2f42999f20badfcc75.
It now shows 97 warnings and 11 errors, where warnings are actually regarding to the code itself, like "assigneed but not used" except for the following two
```
/home/a0795a089/kicad/GerberTools/GerberProjects/GerberProjects.sln:  warning : Don't know how to handle GlobalSection ExtensibilityGlobals, Ignoring.
/home/a0795a089/kicad/GerberTools/GerberProjects/GerberProjects.sln:  warning : Don't know how to handle GlobalSection Performance, Ignoring.
```

and the errors are all like

```
/home/[username]/kicad/GerberTools/GerberProjects/GerberProjects.sln (default targets) ->
(Build target) ->
/home/[username]/kicad/GerberTools/FitBitmapToOutlineAndMerge/FitBitmapToOutlineAndMerge.csproj (default targets) ->
/usr/lib/mono/xbuild/14.0/bin/Microsoft.Common.targets (GenerateResources target) ->

	/usr/lib/mono/xbuild/14.0/bin/Microsoft.Common.targets: error : Tool exited with code: 1. Output: Error: Invalid ResX input.
Position: Line 505, Column 5.
Inner exception: XOR data length expected 2046820352, read 1481
Parameter name: stream
```

I'm totally lost and frustrated. HALP :(

nardev, 2019-05-24:
Still not possible. :(

EuphoricCatface, 2019-05-26:
I moved on to PCBnew standalone mode for panelizing the board.

itavero, 2019-07-07:
Updating the Ionic package, removed the error about the Ionic namespace for me on Mac OS.
Still looking into the TriangleNet stuff ;)

kenyapcomau, 2019-08-02:
Using the prebuilt package GerberTools_2018_8_10.zip

$ mono GerberPanelizer.exe
exe

Unhandled Exception:
System.Reflection.TargetInvocationException: Exception has been thrown by the target of an invocation. ---> System.ArgumentException: XOR data length expected 2046820352, read 1370
Parameter name: stream
  at System.Drawing.Icon.InitFromStreamWithSize (System.IO.Stream stream, System.Int32 width, System.Int32 height) [0x0052e] in <959c23c34b61469a978cd8f9c424f69f>:0 
  at System.Drawing.Icon..ctor (System.Runtime.Serialization.SerializationInfo info, System.Runtime.Serialization.StreamingContext context) [0x00095] in <959c23c34b61469a978cd8f9c424f69f>:0 
  at (wrapper managed-to-native) System.Reflection.MonoCMethod.InternalInvoke(System.Reflection.MonoCMethod,object,object[],System.Exception&)
  at System.Reflection.MonoCMethod.InternalInvoke (System.Object obj, System.Object[] parameters) [0x00002] in <be6fee20a4cf451a8bc9b85c19fe2133>:0 
   --- End of inner exception stack trace ---
  at System.Reflection.MonoCMethod.InternalInvoke (System.Object obj, System.Object[] parameters) [0x00014] in <be6fee20a4cf451a8bc9b85c19fe2133>:0 
  at System.Reflection.MonoCMethod.DoInvoke (System.Object obj, System.Reflection.BindingFlags invokeAttr, System.Reflection.Binder binder, System.Object[] parameters, System.Globalization.CultureInfo culture) [0x0007a] in <be6fee20a4cf451a8bc9b85c19fe2133>:0 
  at System.Reflection.MonoCMethod.Invoke (System.Object obj, System.Reflection.BindingFlags invokeAttr, System.Reflection.Binder binder, System.Object[] parameters, System.Globalization.CultureInfo culture) [0x0002f] in <be6fee20a4cf451a8bc9b85c19fe2133>:0 
  at System.Reflection.MethodBase.Invoke (System.Object obj, System.Object[] parameters) [0x00000] in <be6fee20a4cf451a8bc9b85c19fe2133>:0 
  at System.Reflection.RuntimeConstructorInfo.SerializationInvoke (System.Object target, System.Runtime.Serialization.SerializationInfo info, System.Runtime.Serialization.StreamingContext context) [0x00000] in <be6fee20a4cf451a8bc9b85c19fe2133>:0 
  at System.Runtime.Serialization.ObjectManager.CompleteISerializableObject (System.Object obj, System.Runtime.Serialization.SerializationInfo info, System.Runtime.Serialization.StreamingContext context) [0x00059] in <be6fee20a4cf451a8bc9b85c19fe2133>:0 
  at System.Runtime.Serialization.ObjectManager.FixupSpecialObject (System.Runtime.Serialization.ObjectHolder holder) [0x00088] in <be6fee20a4cf451a8bc9b85c19fe2133>:0 
  at System.Runtime.Serialization.ObjectManager.DoFixups () [0x0005a] in <be6fee20a4cf451a8bc9b85c19fe2133>:0 
  at System.Runtime.Serialization.Formatters.Binary.ObjectReader.Deserialize (System.Runtime.Remoting.Messaging.HeaderHandler handler, System.Runtime.Serialization.Formatters.Binary.__BinaryParser serParser, System.Boolean fCheck, System.Boolean isCrossAppDomain, System.Runtime.Remoting.Messaging.IMethodCallMessage methodCallMessage) [0x00077] in <be6fee20a4cf451a8bc9b85c19fe2133>:0 
  at System.Runtime.Serialization.Formatters.Binary.BinaryFormatter.Deserialize (System.IO.Stream serializationStream, System.Runtime.Remoting.Messaging.HeaderHandler handler, System.Boolean fCheck, System.Boolean isCrossAppDomain, System.Runtime.Remoting.Messaging.IMethodCallMessage methodCallMessage) [0x000a2] in <be6fee20a4cf451a8bc9b85c19fe2133>:0 
  at System.Runtime.Serialization.Formatters.Binary.BinaryFormatter.Deserialize (System.IO.Stream serializationStream, System.Runtime.Remoting.Messaging.HeaderHandler handler, System.Boolean fCheck, System.Runtime.Remoting.Messaging.IMethodCallMessage methodCallMessage) [0x00000] in <be6fee20a4cf451a8bc9b85c19fe2133>:0 
  at System.Runtime.Serialization.Formatters.Binary.BinaryFormatter.Deserialize (System.IO.Stream serializationStream, System.Runtime.Remoting.Messaging.HeaderHandler handler, System.Boolean fCheck) [0x00000] in <be6fee20a4cf451a8bc9b85c19fe2133>:0 
  at System.Runtime.Serialization.Formatters.Binary.BinaryFormatter.Deserialize (System.IO.Stream serializationStream, System.Runtime.Remoting.Messaging.HeaderHandler handler) [0x00000] in <be6fee20a4cf451a8bc9b85c19fe2133>:0 
  at System.Runtime.Serialization.Formatters.Binary.BinaryFormatter.Deserialize (System.IO.Stream serializationStream) [0x00000] in <be6fee20a4cf451a8bc9b85c19fe2133>:0 
  at System.Resources.ResourceReader.DeserializeObject (System.Int32 typeIndex) [0x00019] in <be6fee20a4cf451a8bc9b85c19fe2133>:0 
  at System.Resources.ResourceReader._LoadObjectV2 (System.Int32 pos, System.Resources.ResourceTypeCode& typeCode) [0x0035c] in <be6fee20a4cf451a8bc9b85c19fe2133>:0 
  at System.Resources.ResourceReader.LoadObjectV2 (System.Int32 pos, System.Resources.ResourceTypeCode& typeCode) [0x00000] in <be6fee20a4cf451a8bc9b85c19fe2133>:0 
  at System.Resources.ResourceReader.LoadObject (System.Int32 pos, System.Resources.ResourceTypeCode& typeCode) [0x00022] in <be6fee20a4cf451a8bc9b85c19fe2133>:0 
  at System.Resources.RuntimeResourceSet.GetObject (System.String key, System.Boolean ignoreCase, System.Boolean isString) [0x000be] in <be6fee20a4cf451a8bc9b85c19fe2133>:0 
  at System.Resources.RuntimeResourceSet.GetObject (System.String key, System.Boolean ignoreCase) [0x00000] in <be6fee20a4cf451a8bc9b85c19fe2133>:0 
  at System.Resources.ResourceManager.GetObject (System.String name, System.Globalization.CultureInfo culture, System.Boolean wrapUnmanagedMemStream) [0x00028] in <be6fee20a4cf451a8bc9b85c19fe2133>:0 
[... cut ...]
  at GerberCombinerBuilder.GerberPanelizerParent.InitializeComponent () [0x00c08] in <4c0c6cc0943e4803a6dcff3fea38c2f8>:0 
  at GerberCombinerBuilder.GerberPanelizerParent..ctor () [0x00006] in <4c0c6cc0943e4803a6dcff3fea38c2f8>:0 
  at (wrapper remoting-invoke-with-check) GerberCombinerBuilder.GerberPanelizerParent..ctor()
  at GerberCombinerBuilder.Program.Main () [0x0000b] in <4c0c6cc0943e4803a6dcff3fea38c2f8>:0

No idea what this means. An older build from 2017 from this post https://github.com/ThisIsNotRocketScience/GerberTools/issues/14#issuecomment-344358766 worked.

maludwig, 2020-06-06:
I was able to get commit `95d5c403b1da2338430e5bf0c12f9ea41e87dec3` working. 2019-09-22, so a lot more recent than 2017.

CRImier, 2020-06-06:
did you use the ./build.sh script for that?

trembel, 2020-07-22:
Any solutions to this?

46cv8, 2020-09-26:
> I was able to get commit `95d5c403b1da2338430e5bf0c12f9ea41e87dec3` working. 2019-09-22, so a lot more recent than 2017.

I was able to get close to building it using 95d5c403b1da2338430e5bf0c12f9ea41e87dec3 on ubuntu 20.04 with mono-complete installed but I ran into some errors about a missing class "vec2" which appears to be in the codebase so I don't know why xbuild couldn't find it. In order to get it to properly detect Triangle I had also do the following.

"mv GerberProjects/packages/Triangle.0.0.6-Beta3/ GerberProjects/packages/Triangle.0.0.6/"

Any other specifics about how you actually got this to build would be appreciated as it certainly was a useful tool when I used windows.

dromer, 2020-11-13:
Full build log on debian Buster with mono-xbuild installed:

```
$ ./build.sh 
+ export TERM=xterm
+ TERM=xterm
+ nuget Triangle 0.0.6-Beta3
+ package=Triangle
+ version=0.0.6-Beta3
++ mktemp
+ zip=/tmp/tmp.p4Uleid7dz.zip
+ url=https://www.nuget.org/api/v2/package/Triangle/0.0.6-Beta3
+ dir=GerberProjects/packages/Triangle.0.0.6-Beta3
+ test -d GerberProjects/packages/Triangle.0.0.6-Beta3
+ nuget DotNetZip 1.12.0
+ package=DotNetZip
+ version=1.12.0
++ mktemp
+ zip=/tmp/tmp.aDTRDlSHhN.zip
+ url=https://www.nuget.org/api/v2/package/DotNetZip/1.12.0
+ dir=GerberProjects/packages/DotNetZip.1.12.0
+ test -d GerberProjects/packages/DotNetZip.1.12.0
+ nuget OpenTK 3.0.1
+ package=OpenTK
+ version=3.0.1
++ mktemp
+ zip=/tmp/tmp.lQtaTYrWfk.zip
+ url=https://www.nuget.org/api/v2/package/OpenTK/3.0.1
+ dir=GerberProjects/packages/OpenTK.3.0.1
+ test -d GerberProjects/packages/OpenTK.3.0.1
+ nuget OpenTK.GLControl 3.0.1
+ package=OpenTK.GLControl
+ version=3.0.1
++ mktemp
+ zip=/tmp/tmp.sbTOZ6hgOA.zip
+ url=https://www.nuget.org/api/v2/package/OpenTK.GLControl/3.0.1
+ dir=GerberProjects/packages/OpenTK.GLControl.3.0.1
+ test -d GerberProjects/packages/OpenTK.GLControl.3.0.1
+ nuget netDXF 2.0.2
+ package=netDXF
+ version=2.0.2
++ mktemp
+ zip=/tmp/tmp.FW4sUdzQk2.zip
+ url=https://www.nuget.org/api/v2/package/netDXF/2.0.2
+ dir=GerberProjects/packages/netDXF.2.0.2
+ test -d GerberProjects/packages/netDXF.2.0.2
+ nuget netDXF 0.9.3
+ package=netDXF
+ version=0.9.3
++ mktemp
+ zip=/tmp/tmp.VDDK99C3vv.zip
+ url=https://www.nuget.org/api/v2/package/netDXF/0.9.3
+ dir=GerberProjects/packages/netDXF.0.9.3
+ test -d GerberProjects/packages/netDXF.0.9.3
+ nuget GlmNet 0.5.1
+ package=GlmNet
+ version=0.5.1
++ mktemp
+ zip=/tmp/tmp.xnytCflD5e.zip
+ url=https://www.nuget.org/api/v2/package/GlmNet/0.5.1
+ dir=GerberProjects/packages/GlmNet.0.5.1
+ test -d GerberProjects/packages/GlmNet.0.5.1
+ nuget DockPanelSuite 3.0.6
+ package=DockPanelSuite
+ version=3.0.6
++ mktemp
+ zip=/tmp/tmp.HDpDhxUeNz.zip
+ url=https://www.nuget.org/api/v2/package/DockPanelSuite/3.0.6
+ dir=GerberProjects/packages/DockPanelSuite.3.0.6
+ test -d GerberProjects/packages/DockPanelSuite.3.0.6
[... 251 lines cut ...]
/usr/lib/mono/xbuild/14.0/bin/Microsoft.Common.targets (GetFrameworkPaths target) ->

	/usr/lib/mono/xbuild/14.0/bin/Microsoft.Common.targets:  warning : TargetFrameworkVersion 'v4.8' not supported by this toolset (ToolsVersion: 14.0).

/home/dreamer/Sources/_downloads/GerberTools/GerberProjects/GerberProjects.sln (default targets) ->
(Build target) ->
/home/dreamer/Sources/_downloads/GerberTools/DirtyPCBs/Base64Extractor/DirtyPCB_Base64Extractor.csproj (default targets) ->
/usr/lib/mono/xbuild/14.0/bin/Microsoft.Common.targets (GetReferenceAssemblyPaths target) ->

	/usr/lib/mono/xbuild/14.0/bin/Microsoft.Common.targets:  warning : Unable to find framework corresponding to the target framework moniker '.NETFramework,Version=v4.8'. Framework assembly references will be resolved from the GAC, which might not be the intended behavior.

/home/dreamer/Sources/_downloads/GerberTools/GerberProjects/GerberProjects.sln (default targets) ->
(Build target) ->
/home/dreamer/Sources/_downloads/GerberTools/DirtyPCBs/Base64Extractor/DirtyPCB_Base64Extractor.csproj (default targets) ->
/usr/lib/mono/xbuild/14.0/bin/Microsoft.Common.targets (ResolveAssemblyReferences target) ->

	/usr/lib/mono/xbuild/14.0/bin/Microsoft.Common.targets:  warning : Reference 'System.Data.DataSetExtensions' not resolved
	/usr/lib/mono/xbuild/14.0/bin/Microsoft.Common.targets:  warning : Reference 'Microsoft.CSharp' not resolved
	/usr/lib/mono/xbuild/14.0/bin/Microsoft.Common.targets:  warning : Reference 'System.Net.Http' not resolved

	 16 Warning(s)
	 0 Error(s)

Time Elapsed 00:00:00.7466700
```

cjdg, 2020-12-05:
Using Ubuntu 20.10 and cannot both run or compile

positron96, 2021-01-13:
Hi all, I've managed to build the project in ubuntu. Here are the needed changes: https://github.com/ThisIsNotRocketScience/GerberTools/pull/128/files

kemerelab, 2021-01-13:
Thanks @positron96 - I just cloned your repository and was able to compile in Ubuntu 20.04!

CRImier, 2021-01-13:
@positron96 @kemerelab build - yes, but how about running, too?

kemerelab, 2021-01-13:
I'm a new user, so that's harder for me to assess. My first test project failed to produce all the expected layers, but I think that may be a failure of the naming of my initial files.

kemerelab, 2021-01-14:
Notes on running with @positron96's changes:

 - Selecting Help/Workflow in the Help menu causes the program to terminate.
 - Most importantly, the hierarchical view menu and controls on the right hand side are not there. (I only see the PCB)

positron96, 2021-01-14:
As noted here: https://github.com/ThisIsNotRocketScience/GerberTools/issues/71#issuecomment-455454543, the right side panel might be unfixable since the library is windows-only.

taotieren, 2021-04-26:
Winelib is a development toolkit which allows you to compile your Windows applications on Unix.

https://wiki.winehq.org/Winelib_User's_Guide

CRImier, 2021-04-26:
@taotieren If it weren't for you sharing a free and open-source tool, I'd be 100% certain your comment is spam. Having said that, thank you for your suggestion!

taotieren, 2021-04-26:
Sorry, I prefer gerbertool to support the Linux platform.