ThisIsNotRocketScience/GerberTools #85: Silkscreen Images with gaps get filled up

rupin, 2019-04-24:
Hello,
I designed a board in EASYEDA using an image on the top silkscreen. Notice how the yellow silkscreens ( Text BlinkyBadge and the Instagram Logo) have a gap in them. 

[image]

The gap vanishes in the silkscreen when the pcb is panelised in the paneliser. 

[image]

Is this by intent or is there a switch in the software that allows to change this behaviour?

witnessmenow, 2019-06-18:
I'm after running into the same issue

indivdual
[image]

After panelizing, 
[image]

OshPark wont even load the file, 

JLCPCB shows the same as the image generated by the panelizer
[image]

StijnKuipers, 2019-09-23:
Polygon winding is quite hard in a format that doesnt officially support closed polygons correctly - the gerbers are probably fine, but prerendering without following the actual photoprocess of the production is quite hard to get right. 

We'll get there eventually :-)

If you attach some gerbers, I can take a look

pjsg, 2020-06-08:
I have the same (or very similar issue). I wondered if it was just the gerber viewer, but the board that came back from the board house (jlcpcb) had the same issue:

[image]

[RearPanel.Zip](https://github.com/ThisIsNotRocketScience/GerberTools/files/4743550/RearPanel.Zip)

I had added two boards in panelizer (they were very similar -- the front and back panel) and closed loops on the lettering got filled in. I will admit that the gerber viewers showed them as being filled in before I sent them off. Happily these are just panels and have no circuitry in them.

positron96, 2020-10-14:
Same issue here. I have a custom font, and Os and As are filled. I could live with that, but an explanation text on the board (small configuration manual) has icons that become unreadable

positron96, 2021-01-14:
Hi guys. This seems to be fixed in the current master (you need to compile yourselves). I've tried to process my board again with latest source, and the results are much better than they were in October (I used July release back then).