ThisIsNotRocketScience/GerberTools #91: Spurious extra circular tracks on inner layers

Gasman2014, 2019-05-31:
Using this to panellise some Gerbers created with Kicad using  version 1.0.6345.32009 on W7 running on a VirtualBox. The gerber files include some circular tracks. The gerbers look correct (in two different viewers) prior to panelisation, however, after panelisation, weird circular artefacts on the two inner copper layer are apparent. Initially the circular cutout was defined as a circle but I have replaced this and all other arcs of >180º with arcs of < 180º as I believe that circular gerber tracks have been problematic in certain situations.
My primary pc is a mac so am running this in virtualisation but don't think this should be an issue. I am prepared to try building this from a newer version if this will address the issue but I am unclear if it will help. A newer binary edition would be very helpful. 

After panelisation:
[image]

Gerber view in 'Cuprum'
[image]

Gerber view in 'gerbview'
[image]

Gasman2014, 2019-05-31:
For completeness, I ensured .gm1 was renamed to .gko, used protel extensions and added and used an auxiliary axis to ensure the source Gerbers were compliant with GerberPanelizer. I am happy to share the source and output gerbers if that is of any help.

bzroom, 2021-01-01:
@Gasman2014 , did you ever find a solution or workaround to gerber panelizer?

Gasman2014, 2021-01-03:
This was only intended as a small batch run but thought I could get more out of the batch if I panelized them myself. In the end I just got the fab to panelize them for me.
I have used this software successfully before but I would really prefer a macOS solution. I have used a java app with Kicad which works OK but can't put my finger on it right now.

closed 2021-01-03 by Gasman2014

StijnKuipers, 2021-01-03:
hmm missed this one - any chance on getting some gerber files to check with?

bzroom, 2021-01-03:
Please see my issue here:
https://github.com/ThisIsNotRocketScience/GerberTools/issues/108

I attached gerbers there, I think it's a similar problem.

I retried with my new board and the latest version of Panelizer. The result
was different but still broken. It seems like copper fill areas do not
respect if there is a hole inside the copper fill. In my example gebers,
there is a drill hole in the board outline, but the copper area fills over
the whole area. The copper area also fills over traces that are inside the
area. It's not subtracting islands for board holes or even internal traces
within the profile.

Thank you!

bzroom, 2021-01-03:
[image: image.png]

Gasman2014, 2021-01-03:
I would be happy to share the original gerbers but need to look them out. Had to replace my HDD recently and have not yet reinstalled any virtualisation stuff so can't test them easily but it was a fairly consistent problem with a number of similar designs of different sized round tracks.