ThisIsNotRocketScience/GerberTools #92: Cannot add gerber files with "Add" button

qqzwc, 2019-07-02:
For more information about calling real-time (JIT) debugging instead of this dialog box,

See the end of this message.
Abnormal text
   System.NotImplementedException: This method or operation is not implemented.
   At GerberViewer.LayerList.AddGerberFile (Object sender, EventArgs e)
   Windows.Forms.Control.OnClick (EventArgs e)
   Windows.Forms.Button.OnClick (EventArgs e)
   At System.Windows.Forms.Button.OnMouseUp
   At System.Windows.Forms.Control.WmMouseUp (Message sm, MouseButtons Button, Int32 clicks)
   At System.Windows.Forms.Control.WndProc (Message-m)
   At System.Windows.Forms.ButtonBase.WndProc (Message-m)
   At System.Windows.Forms.Button.WndProc (Message-m)
   At System.Windows.Forms.Control.ControlNativeWindows.OnMessage (Message s m)
   At System.Windows.Forms.Control.ControlNativeWindows.WndProc (Message s m)

At System.Windows.Forms.Windows.Window.Callback (IntPtPtr hWnd, Int32 msg, IntPtr wparam, IntPtr lparam)

Yourigh, 2019-08-02:
I confirm, same here:
[image]
App was run as admin. Path was not long.

cowboy, 2020-10-01:
Same issue, with GerberTools_2020_7_30. I literally just downloaded the zip, ran GerberViewer.exe, and clicked Add. Haven't tried anything else yet.