ThisIsNotRocketScience/GerberTools #93: Wont work with new vesion of Eagle

StanislavJochman, 2019-08-04:
Hi, 
I recently discovered that Panelizer would not work with the new version of Eagle. New eagle changed the way how they name their files and Panelizer would not read them. After renaming files back to old names everything works so I created this python script which does it automaticaly for you. It uses OS functions so it should work on all operating systems.

https://github.com/StanislavJochman/GerberTools_Eagle_Fix

Best regards Stanley

rahmanshaber, 2019-09-10:
I am also facing the issue.
Thanks @StanislavJochman  for your script. 
Let's see how pcb look in real life.

StijnKuipers, 2019-10-01:
New eagle no longer allows you to set up your own job-settings, filenames & target paths?

StanislavJochman, 2019-10-01:
hello,
i dont know if you can change it but rs274x is commonly used in pcbs
factory and its set es default in eagle so I made this converter which
converts filenames to correct ones.

------------------------------------------------
S pozdravom
StanislavJochman
www.stanislavjochman.sk
Ceo at www.robotcing.sk

TimFFischer, 2019-11-17:
Thanks @StanislavJochman:
I got inspired and wrote a batch file in order to automatically convert multiple BRD-files to GERBER-files for GerberTools:
[https://wiki.mexle.hs-heilbronn.de/lib/exe/fetch.php?media=convert_brd_to_gerber.rar](url)

Converts all BRD-files in folder "BRD" to GERBER-files in folder "GERBER".
Needs EAGLE 8.6 or higher. Details in readme

StanislavJochman, 2019-11-17:
@TimFFischer link doesn't work unfortunately.

Best regards StanislavJochman

pierow2k, 2020-07-28:
> @TimFFischer link doesn't work unfortunately.

@StanislavJochman - The URL works but the link points to the wrong location.  I was able to retrieve the @TimFFischer batch file by copying the URL text and pasting it into a browser.  I think this link should work:

[https://wiki.mexle.hs-heilbronn.de/lib/exe/fetch.php?media=convert_brd_to_gerber.rar](https://wiki.mexle.hs-heilbronn.de/lib/exe/fetch.php?media=convert_brd_to_gerber.rar)

StanislavJochman, 2020-07-28:
Ok thanks, I will post it here if someone wants it: https://wiki.mexle.hs-heilbronn.de/lib/exe/fetch.php?media=convert_brd_to_gerber.rar

closed 2020-07-28 by StanislavJochman

Mikegyver, 2021-03-10:
> Hi,
> I recently discovered that Panelizer would not work with the new version of Eagle. New eagle changed the way how they name their files and Panelizer would not read them. After renaming files [...]

hi @StanislavJochman i think your python script did not rename back solderpaste bottom to old name. here is my output when i executed the python script.

[image]

hope this could be fix soon. best regards.

StanislavJochman, 2021-03-10:
> hi @StanislavJochman i think your python script did not rename back solderpaste bottom to old name. here is my output when i executed the python script.
>
> [image]
>
> hope this could be fix soon. best [...]

Hi to what name should it be renamed? I did not use the Gerber panelizer for a long time. I can fix that soon if you can tell me to what it should be renamed to.

Mikegyver, 2021-03-10:
> Hi to what name should it be renamed? I did not use the Gerber panelizer for a long time. I can fix that soon if you can tell me to what it should be renamed to.

it should be rename to .gbp

StanislavJochman, 2021-03-10:
> .gbp

I hope I fix it you can try to redownload the new script and I would be happy to know if it worked.

Best regards Staney Jochman