ThisIsNotRocketScience/GerberTools #96: More than 2 layers

fjullien, 2019-11-07:
I may have misunderstand something but does GerberTools/GerberPanelizer supports more than 2 copper layers ?

navinyate, 2019-11-07:
Same Question here

StijnKuipers, 2019-11-09:
The merger is layer agnostic - it will just merge layers of the same filetype together - by file extention. 

Only drill and outline are special so far - 
Outline it needs for obvious reasons - and special via layers for blind vias is not something I've ever had an opportunity to try - drills are all "drilled through" as far as the merger is concerned right now.

navinyate, 2019-11-09:
But if i pull in the Gerber files which are exported by eagle will not show
board outline.

fjullien, 2019-11-09:
I didn't use protel files extension.I now have *.g2 and *.g3 as internal layers and it is handled separately from top and bottom layers. However, I still not get a good result. My board outline is not a simple shape. I can pm you the gerber files if you want to give it a try.

AndyW999, 2020-05-22:
I have just tried to use this with a 4 layer board done in KiCad and I only get two layers.

Any ideas?