ThisIsNotRocketScience/GerberTools #98: Eagle 9.2.2 gerbers not recognized

Kralik01, 2020-01-01:
Hi,
I recently wanted to merge some gerbers, generated with eagle 9.2.2. The gerbers are RS-274X. I was using your [August 2018 Build](https://github.com/ThisIsNotRocketScience/GerberTools/releases/tag/2018-08-10) (Cause i wasn't been able to compile the newer one, hoewer i tried.) And it just does nothing when i import my files. I saw [This issue](https://github.com/ThisIsNotRocketScience/GerberTools/issues/93) posted by StanislavJochman, but renaming the files won't work for me. Is it fixed in newer version? Is there any solution?
Thanks.

rybitski, 2020-02-07:
I'm experiencing the same issue.

xeroe, 2020-07-13:
using another cam processor template worked for me
(open cam processor in eagle, choose another template in the top left, third party - seed studio 2 layer for example)

MS71, 2020-08-06:
Hi,
i have the same issue with Eagle 9.6.2.
I can't load my gerber files.
Using the seed studio cam processor, doesn't help.
Any hint howto process latest eagle gerber files?

[esp32_module.zip](https://github.com/ThisIsNotRocketScience/GerberTools/files/5034914/esp32_module.zip)

Thore-Krug, 2020-12-31:
@MS71  I create a Modified CAM Version and it works sucessfuly with Eagle 9.6.2
But i have a Issue that there is not a Top Silkscreen on most Boards this seems to be a problem with Gerber Panelizer though.  I created a new Issue for that though https://github.com/ThisIsNotRocketScience/GerberTools/issues/127 

But you can try my .CAM File and see if that works for you.  

[CAM-Gerber-Panelizer.zip](https://github.com/ThisIsNotRocketScience/GerberTools/files/5757682/CAM-Gerber-Panelizer.zip)