ThisIsNotRocketScience/GerberTools #100: Feature Req: Pick and Place file panelizer

marrobHD, 2020-01-04:
Please add a Pick and Place file panelizer

ghoeffner, 2020-02-28:
This feature would be great to have.

CRImier, 2020-03-08:
If you're using KiCad, check this out: https://github.com/CRImier/Panelizer2PnP If you're not using KiCad, pull requests welcome!

rorosaurus, 2020-07-31:
+1, would love to have this feature!

StijnKuipers, 2020-08-05:
There are several versions of a bom loader/panelizer already in the sourcecode - but the feature is not quite advertised since there is not really a "standard" of sorts to be used for this

if somebody can figure out the best crosssection of fileformats to use and can make a decent case for them - I'll add them.

also -> what to do with refdes renaming? if you have only the same panel 10x, you can keep 1 name (R1 on all boards) - but as soon as you merge boards - you need to rename to avoid collissions 

Ive done this for my own usecase - but I can imagine people dont like it when the board says R1 in the silk, but the Bom/PnP refers to A_R1 or something - or R2612 

Any thoughts?

robstoddard, 2021-09-13:
Thoughts:
1)   Rabbit hole?-  a format configurator for PnP file format.    There may be different formats, but if they're reasonably similar, then being able to specify in a dialog things like delimiter and what-goes-where might be sufficient.
2)   Cheap and easy? -   A JSON file or similar output that allows us to pick up the offset coordinates of each board so we may script our own PnP solutions.

I suspect the #2 solution is the best all-around solution here because we, as board designers, have our tools outputting format A and our machines want format B,   so it's highly likely we're doing some script magic *anyway* so throw us a bone, let us run that extra mile and have our scripts panelize the PnP stuff according to the output of your tool.