ThisIsNotRocketScience/GerberTools #103: Additional Milling added (makes a triangle from an curved line)

ranseyer, 2020-02-07:
The gerbertools add an additional line from the beginning bi to the end of the line to a curved milling.
The original line has a width of 50 mil, and the added line (width 0mil) creates an area similar to a triangle.

I woul be happy abou an help or an hint...

[merged-defect.zip](https://github.com/ThisIsNotRocketScience/GerberTools/files/4172315/merged-defect.zip)

[image]

[Original-Gerber-HM-Relais4-flach-v01-034-eagle7.zip](https://github.com/ThisIsNotRocketScience/GerberTools/files/4172324/Original-Gerber-HM-Relais4-flach-v01-034-eagle7.zip)

StijnKuipers, 2020-02-08:
Could you also post the original file? Can't say much about the error on
the combined file alone.

ranseyer, 2020-02-08:
Yes, it is attached above as: Original-Gerber-HM-Relais4-flach-v01-034-eagle7.zip

ranseyer, 2020-02-18:
I can attach it again i needed.

StijnKuipers, 2020-02-18:
Oh right - github mail only showed 1 of 2 links

StijnKuipers, 2020-02-18:
Your GML file has a random unclosed line in the middle - fix that in your eagle file and reexport.

StijnKuipers, 2020-02-18:
[image]

closed 2020-02-18 by StijnKuipers

ranseyer, 2020-02-18:
Hmm, this is on the milling layer. 

In the past i ordered exactly in this way PCBs and the producers milled it exactly as wide as in eagle designed.

But i learned the i can redraw this with an line wideness of 0mil ... 
Thanks for the feedback !

StijnKuipers, 2020-02-19:
a lot of boardhouses have teams of people manually fixing boards without
telling you about this... "magic happens"

These gerber tools dont have the luxury of interpretation :-(