ThisIsNotRocketScience/GerberTools #110: Unhandled exception when trying to import Gerbers to panelizer

milindmm97, 2020-05-12:
[image]
Kindly let me know how to solve this or what I am doing wrong. Runnging the executable available on the release (GerberTools_2019_9_20.zip) 
My system is running windows 10

nevetts, 2021-06-29:
I also have this problem both when trying to add a gerber file and when attempting to export a merged gerber.

robstoddard, 2021-09-13:
I don't know if it is THIS problem, I just know that I had a crash when I tried to export Gerbers without having saved prior to doing so.   Now, I save prior to doing so (expecting it to crash) but it doesn't crash.    I am wondering if this crash has to do with whether the gerberset file has been created or not.   When I get more time to explore, I'll see if I can drive down this exception more, see if it's the same one.