ThisIsNotRocketScience/GerberTools #111: Mixed up output and pcb disappear when rotating

maxholgasson, 2020-06-28:
Hi,

using win 10 with latest updates my merged gerber files are totaly mixed up. Also I can't rotate a instance..it just dissapears when trying to do this

[image]
[image]

StijnKuipers, 2020-06-29:
The rotation happens around the origin of the board - are the boards
centered around 0,0 before you export?
Also - maybe there is too much extra stuff on the GKO layer - there need to
be clean hairline outlines on the GKO for the polygon merger to work

maxholgasson, 2020-06-29:
I'm not exactly sure as I don't know how I can center them/set the point precisely) in kicad.  GKO(=edge?) layer is just a plain rectangle