ThisIsNotRocketScience/GerberTools #112: BreakTab

iharshadec, 2020-07-22:
Hi,

I want to know if there is any way we can change the Breaktab length? Right now If I insert breaktab manually it is of approx 3.5mm but my PCB manufacturer accepts no less than 4mm. 
Q:- Do we have any way to have this changed as per our need?

Also I have notice **Create Breaktabs** works on some PCBs only.

StijnKuipers, 2020-07-28:
You can change the radius of each breaktab.

iharshadec, 2020-07-29:
> You can change the radius of each breaktab.
> […](#)

How to do that? But still breaktab length min 4 is required.

Rixxxi, 2023-07-26:
Changing the radius of a break tab is done by
1) select a certain break tab in the main window
2) change it's radius on the right hand properties
[image]