ThisIsNotRocketScience/GerberTools #113: Merged Gerbers fail to upload to PCB sites

abishur, 2020-08-07:
After making the merged PCB I zip the gerber files and try to upload to Oshpark and JLCPCB.  On Oshpark it fails completely and on JLCPCB it will upload but will show nothing in the gerber preview area.  I'm using Windows 7 and the latest release (1.0.7516.24197 built on 07/30/2020)

iharshadec, 2020-08-26:
Me too have experience same issue ! Something is wrong in generation !

Thore-Krug, 2020-12-30:
Which PCB CAD Tools are you using ?

Gautamjhawar, 2021-02-20:
can you add pictures? I check my files in a Gerber viewer before sending it to my fab house and the pcbs come out perfect

DoganM95, 2023-11-18:
Just tried the panelizer and experienced this issue. Here is some info:
- Using autodesk EAGLE
- Using JLCPCB for gerber upload

My eagle generated gerber:
[mic_sensor_arduino_v0.9_2023-11-18.zip](https://github.com/ThisIsNotRocketScience/GerberTools/files/13400558/mic_sensor_arduino_v0.9_2023-11-18.zip)

Looks like this in eagle preview:
[image]

And like this when adding 2 of then to the panelizer (added a tab between them):
[image]

Panelizer generates this (zipped up all the files it spit out):
[mic_panelized.zip](https://github.com/ThisIsNotRocketScience/GerberTools/files/13400578/mic_panelized.zip)

And looks like this after an upload to jlcpcb:
[image]

Here is a closer look:
[image]