ThisIsNotRocketScience/GerberTools #122: Ask for help:I tried to implement the SetThermal function

gezhaowan, 2020-11-18:
I tried to implement the SetThermal function and tried many methods, but the output result is always incorrect.
I confirm that there is no problem with my drawing logic;
The closest output is like this.
[image]

Do you have any suggestions? @StijnKuipers

` 
internal void SetThermal(double Xoff, double Yoff, double OuterDiameter, double InnerDiameter, double GapWidth, double Rotation)
        {
           Shape.Vertices.Clear();
            GerberApertureType sector1 = new GerberApertureType();
            GerberApertureType sector2 = new GerberApertureType();
            GerberApertureType sector3 = new GerberApertureType();
            GerberApertureType sector4 = new GerberApertureType();
            int sides = (int)Math.Floor(10 * Math.Max(4.0, OuterDiameter / 2));
            {
                double radius = OuterDiameter / 2;
                double padd = 0;// -rotation * Math.PI * 2.0 / 360;
                for (int i = 0; i < sides; i++)
                {
                    double P = i / (double)sides * Math.PI * 2.0 + padd;
                    double x = (double)(Xoff + Math.Sin(P) * radius);
                    double y = (double)(Yoff + Math.Cos(P) * radius);
                    if ((y < Yoff - GapWidth / 2) && (x < Xoff - GapWidth / 2))
                    {
                        sector1.Shape.Add(x, y);
                    }
                    if ((y < Yoff - GapWidth / 2)&& (x >= Xoff + GapWidth / 2))
                    {
                        sector2.Shape.Add(x, y);
                    }
                    if ((y >= Yoff + GapWidth / 2) && (x < Xoff - GapWidth / 2))
                    {
                        sector3.Shape.Add(x, y);
                    }
                    if ((y >= Yoff + GapWidth / 2) && (x >= Xoff + GapWidth / 2))
                        sector4.Shape.Add(x, y);
                }
            }

            {
                double radius = InnerDiameter / 2;
                //int sides = (int)Math.Floor(10 * Math.Max(2.0, radius));
                double padd = 0;// -rotation * Math.PI * 2.0 / 360;
                Shape.Vertices.Clear();
                for (int i = sides-1; i >= 0; i--)
                {
                    double P = i / (double)sides * Math.PI * 2.0 + padd;
                    double x = (double)(Xoff + Math.Sin(P) * radius);
                    double y = (double)(Yoff + Math.Cos(P) * radius);
                    if ((y < Yoff - GapWidth / 2) && (x < Xoff - GapWidth / 2))
                    {
                        sector1.Shape.Add(x, y);
                    }
                    if ((y < Yoff - GapWidth / 2) && (x >= Xoff + GapWidth / 2))
                    {
                        sector2.Shape.Add(x, y);
                    }
                    if ((y >= Yoff + GapWidth / 2) && (x < Xoff - GapWidth / 2))
                    {
                        sector3.Shape.Add(x, y);
                    }
                    if ((y >= Yoff + GapWidth / 2) && (x >= Xoff + GapWidth / 2))
                        sector4.Shape.Add(x, y);
                }
            }
            sector1.Shape.Close();
            sector2.Shape.Close();
            sector3.Shape.Close();
            sector4.Shape.Close();
            Parts.Add(sector1);
            Parts.Add(sector2);
            Parts.Add(sector3);
            Parts.Add(sector4);
}
`

StijnKuipers, 2020-11-18:
Could you send me a minimal testfile with a bunch of different-sized
thermals?

gezhaowan, 2020-11-19:
[PCB Benchmark.zip](https://github.com/ThisIsNotRocketScience/GerberTools/files/5563749/PCB.Benchmark.zip)
This is the test file I used. There is only one thermal pad inside.

gezhaowan, 2020-11-19:
[image]
I have solved it. The main reason is that PolyLine is initialized with 4 points. After clearing it, it is fine.

closed 2020-11-19 by gezhaowan

StijnKuipers, 2020-11-19:
Cool!
Thanks!
:-D