ThisIsNotRocketScience/GerberTools #124: Only one grb file generated

jpduplessis1986, 2020-12-03:
First of its my first time posting a issue on github.  So I hope it is ok, I first went through all the issues to see if there is not something similar.  I used the panalizer tool and as soon as I export it only generates four files.  grb,txt,xln,gko.  As well as the two images.  So the file shows drill holes and a grey background.  If I open it in a online greber viewer it shows the same.  But when I open it with your viewer the greber file shows all the silk and tracks on one file.  I used eagle cad to generate my original files.  I am sure if it is only one file it will not be manufactured correctly.  If you can assist with some information, it will be well appreciated
[image]
.  Thank you

jpduplessis1986, 2020-12-03:
Viewing the file in your viewer.
[image]

jpduplessis1986, 2020-12-03:
[image]

StijnKuipers, 2020-12-03:
The tool merges files by extentions if it cant determine the layers - what are your filenames like?

If you follow the "chinese convention" standardnames (gbo/gbl/gto/gbl etc) it should merge separately

zenandgary, 2021-01-12:
Make sure you check the following option in KiCad when exporting your Gerber files otherwise you will end up with the greyed out images.
[image]

Gautamjhawar, 2021-02-20:
I was using eagles job files and encountered the same problem, use the python script linked below to convert your .gbr files to the "chinese format" (for two layers only for now, it uses the file names to change the extension, and leaves anything else in the folder unchanged)
https://github.com/Gautamjhawar/Gerber-extension-changer