ThisIsNotRocketScience/GerberTools #125: Gerber Panelizer not displaying any gerber file contents on Windows 10 64 bit (Intel gpu).

lukolszewski, 2020-12-26:
Hi,

I'm trying the app for the first time so it is possible I'm using it wrong. However I saw vids/read articles how it is supposed to work and I'm pretty sure this is a bug. 
I downloaded official gerber test files from https://www.ucamco.com/en/gerber/downloads  (The Gerber File Format - Test files (zip) from 15/10/2020) and placed one file named 2-13-1_Two_square_boxes.gbr in a folder called test1.
I opened the app (on Windows 10 Pro , 64bit, Intel HD 5500 driver version 20.19.15.5126), clicked File->New. Then dragged the folder onto the middle of the document.  The document didn't change at all. The Board tree on the right top changed to include  test1 under Gerbers. I can right-click on it and choose add instance. The instance always gets added with coordinates of -4 and 113.98 (panel size 100x100). If I change the coordinates manually to for example 20x20 and click View->Zoom to fit I manage to get a small orange cross on the document. Nothing else. This is shown below:
[image]

lukolszewski, 2020-12-26:
Also, I tested it with official gerber test files after it didn't work with my gerbers. It behaves the same for any gerber file I try. I also tried re-downloading the app and a much older version I found a download link for somewhere. The results are the same.

midvok, 2021-01-02:
I have the same problem. I import the Gerber files (generated from EAGLE) and it just displays the cross hair, but no PCB outline is visible. The program doesn't even trough any exception or warning, so I have no clue if the data were parsed correctly or not.

Thore-Krug, 2021-01-03:
@midvok  You need to adjust the CAM output. The program just merges any file with the same extension. you need to export in a legacy format for it to work. I created a CAM file which sort of works. Some PCBs have no trouble on some the silkscreen is missing ( Still looking into it ) but creating panels work flawlessly.  So if you want you can tinker around with it. 

I attached the CAM File: 
[CAM-Gerber-Panelizer.zip](https://github.com/ThisIsNotRocketScience/GerberTools/files/5761399/CAM-Gerber-Panelizer.zip)

derpicknicker1, 2021-05-12:
@Thore-Krug 
Thank you for the Cam file. Finally i got it working.
Did you manage to solve the silkscreen issue? It's missing on all of my bords after panelizing.

Thore-Krug, 2021-05-16:
@derpicknicker1  I had some Boards which had no silkscreen, but it worked on others.  Due to cost reasons I switched my PCB CAD toolage to Kicad so I didnt Investigated it further.