ThisIsNotRocketScience/GerberTools #126: output of panelizer is completly grey

Lopfi, 2020-12-30:
Hi, im sorry if Im just doing something wrong because I have never done this but when I export my panelized pcbs the image is almost completely grey and even in a gerber-viewer it shows exactly the same.
[image]

Thore-Krug, 2020-12-30:
Hi, 

I wanted to ask which CAD Tool you were using. I used Eagle 9 and ran into the same problem but i was tinkering around today and create a cam file which worked and produced the desired output. I havent ordered it but at first glance it looks all OK.

Lopfi, 2020-12-30:
I used fusion 360 electronics

Thore-Krug, 2020-12-30:
Im not familiar with Fusion 360 Electronic but since you have a eagle subscription included in fusion360 i would just go over to eagle ( You can link them together ) And use the .cam file i created to export the PCB. For Some reason it exported the Top silkscreen on the first Board fine but not on 2 other Boards I tested thats something i have to take a look into. 

Worked on the first board i tried it with. 

but not on the second, even though i havent changed anything. 
This seems to be a Problem with the Gerber Panelizer though as the direct CAM Output has the Silkscreen on it. 

I attach you the .cam file for you to try out. 

PCBs without top silkscreen are still better than no pcbs ;) 

[CAM-Gerber-Panelizer.zip](https://github.com/ThisIsNotRocketScience/GerberTools/files/5755951/CAM-Gerber-Panelizer.zip)

Lopfi, 2021-01-03:
Thanks a lot for the help I didn't use eagle but instead imported the cam into Fusion360 and it worked with all my PCBs but even after trying around I couldn't get the  silkscreen to work.

Thore-Krug, 2021-01-03:
Yeah im still figuring out why this occurs. It worked once but not on other PCBs. The PCB that worked had no through hole components..  I have to try with another SMD only Board.

Thore-Krug, 2021-01-05:
So i saw the the Author of this tool was writing about using Dirty PCBs ( A PCB Fab ) for his Projects. I used the CAM file they provide and it worked including the silkscreen. There is a warning from Eagle about a possible misalignment as this CAM uses a mix of old and new Formats and ignores the PCB STackup ( Which is unimportant though as we dont have controlled impedance with 2 layers which this tool only supports asfaik anyway ). 

I just threw a general look over it in ther Gerber Viwer and there isnt something i noticed that is completely wrong. 

Seems like the Author optimized for his usecase :-D 

Heres the CAM File: 
[dirt_cheap_dirty_boards.v1.1.cam.zip](https://github.com/ThisIsNotRocketScience/GerberTools/files/5768713/dirt_cheap_dirty_boards.v1.1.cam.zip)

zenandgary, 2021-01-12:
If you're using KiCad make sure you export your Gerber files with the following option checked otherwise you won't generate the files that Panelizer expects.
[image]

MayckonJhonatas, 2022-01-22:
hi @zenandgary, i'm having the same problem. i'm using eagle.

CRImier, 2022-09-05:
The latest downloadable build (2020) cannot process gerbers from KiCad 6, too - unless you disable aperture macros when plotting the gerbers. If you don't, it might result in exactly this kind of outcome as shown in the OP picture!

Alternatively: install visual studio community edition with desktop C# support, download this repo as a ZIP, open the project as described in README, open the Panelizer subproject, then just press "Start", it will compile you a new Panelizer with proper support for things.