ThisIsNotRocketScience/GerberTools #127: Gerber Panelizer does not merge Silkscreen on Some Boards

Thore-Krug, 2020-12-31:
I created a CAM File for the Export of Eagle since the Standard Eagle Gerber Output now uses the same file ending for all Files. 

**It outputs the following Files files:**

- Bottom Paste ( bCream ) in a .GBP file 
- Top Paste ( tCream) in a GTP file 
- Bottom Solder Mask ( bStop) in a .GBS File
- Top Solder Mask (tStop) in a .GTS File
- Bottom Silkscreen (bPlace/bNames) in a .GBO File
- Top Silkscreen (tPlace/bNames) in a .GTo File
- Bottom Copper (Bottom/Pads/Vias) in a .GBL File
- Top Copper (Top/Pads/Vias) in a .GTL File 
- Milling ( Milling) into a .GML File 
- Outline ( Dimension) into a .GKO File 
- Drill File gets exported as .TXT File 

This all gets zipped up into a .zip archive by the CAM Processor. 

A Tree of the File structure within the zip File: 

````
Project-Gerber/
├── Gerber/
│   ├── gerber_job.gbrjob
│   ├── other.GML
│   └── other.GKO
│ 
├── Project.GBL
├── Project.GBO
├── Project.GBP
├── Project.GBS
├── Project.GTL
├── Project.GTO
├── Project.GTP
├── Project.GTS
└──Project.TXT

````

Importing into the Panelizer works flawlessly with this Setup and i produce a panelized Gerber every Time. 
Yet the top Silkscrren is missing on every Board exept one i tried to panelize. 

Which is very mcuh to my surprise as all Boards used the same CAM Output as the Board which had no issues with the Top Silkscreen. 

I checked the Gerber Output of my CAD Program ( Eagle ) to ensure that the top Layer is present which is in Fact the case and it can also be read from other Gerber Viewers. 

[image]

On the Left is the Gerber Which has the Top Silkscreen Issue, The Silkscreen is present even after panelization on the board located on the right. 

And even though both Files contain the Top Silkscreen Layer its only present in one File after creating the Panel. 
I tried multiple other Boards but no sucess i re exported the Boards to make sure its not a broken file or similar. 

But on the non sucessful Board the Top Silkscreen gets not merged. 

( Sorry for the Phone Fotos i tried to make screenshots at the right moment but never were able to time it correctly ) 

**Sucessful Board**

[image]

**Board with Silkscreen Error**

[image]

Hence there is no merged Silkscreen Output for the Top Silkscreen. 

**Bottom**

[image]

**Top**

[image]

noisemaker00, 2021-02-09:
I downloaded the source code and debugging it.
In my case the board has no top silkscreen and I noticed the *.GTO file does not exist in the `GeneratedFiles` list.
The program crash in `GerberPanel.cs` on `GIC2.ClipBoard(f, tempfile, Logger);` and an exception is thrown, so the layer file is not processed.

Going deeper the exception is throw in `ImageCreator` on
```
ClipperLib.Clipper CP = new ClipperLib.Clipper();
foreach (var ol in ols)
{
    ...
    var R = ol.FindLargestPolygon();
    var poly = R.Item2.toPolygon();
    ...
}
```
due to R is null
when this happens `ols` has 2 items:
```
Layer Outline (.GKO)
Layer Mill(.GML)
```
and crash on `ol ` layer is `BoardLayer.Mill`

I'm keep investigatinv but I've not knowledge on how gerber works in details, so I hope these information may be usefull to the developer to fix.

StijnKuipers, 2021-02-09:
If you have a zip with the gerbers, I can take a look..

noisemaker00, 2021-02-09:
Yes, I'm using this one
[esp8266 breakout board_2021-02-08.zip](https://github.com/ThisIsNotRocketScience/GerberTools/files/5952962/esp8266.breakout.board_2021-02-08.zip)

noisemaker00, 2021-02-09:
I tried to add a check wrapping after R is assigned
```
var R = ol.FindLargestPolygon();
if (R != null)
{
   var poly = R.Item2.toPolygon();
   ...
   CP.AddPolygon(poly, ClipperLib.PolyType.ptClip);
}
```
and the top silk screen has been added on the final merged board image, but I don't know if something may me missing due to the skipped code in curly braces

StijnKuipers, 2021-02-10:
that should be safe - could you commit that change?

noisemaker00, 2021-02-15:
I'm still having crashes on the merged files when I open with the viewer, so I suspect something wrong happens during the process even if the null exception seems resolved

weissjuergen, 2022-01-10:
Since ths is a while ago and I may have missed the point. Is there any solution to be able to panalize Eagle 9.6 output ?

CRImier, 2022-08-29:
do you have the "clip to board outlines" checkbox checked in panel properties? uncheck that and regenerate. it's an option enabled by default that's supposed to clip silk and I assume other elements to board outline, but instead seems to remove most if not all of the silk.

CRImier, 2022-09-05:
The latest downloadable build (2020) cannot process gerbers from KiCad 6, too - unless you disable aperture macros when plotting the gerbers. Alternatively: install visual studio community edition with desktop C# support, download this repo as a ZIP, open the project as described in README, open the Panelizer subproject, then just press "Start", it will compile you a new Panelizer with proper support for things.