ThisIsNotRocketScience/GerberTools #129: Some silkscreen is missing after using panelizer tool

noisemaker00, 2021-02-04:
I try to use the panelizer tool to merge togheter different boards but the output does not have some silkscreen.
I used EAGLE to draw

Step to reproduce:
Exported to Gerber using the example CAM you give in another issue
The Gerber Viewer tool of the suit show al PCBs silkscreen correctly (before merging)
Add to Panelizer Tool succesfully
Aftert merge some silkscreen is missing

Gautamjhawar, 2021-02-20:
Is it possible for you to attach a picture? i am using eagle(9.6.2) too and it works fine for me

ckemere, 2021-03-07:
I have the same experience. I believe it happens when the silkscreen layer extends outside the bounds of the board outline.

VinnyCordeiro, 2021-03-08:
I'm having the same issue, using both Eagle 7.6 and KiCad 5.1.9: I'm trying to merge 3 boards (one of them with 2 instances), only one of the boards gets the silkscreen properly. All of the PCBs are 4 layers. I'm attaching the files for inspection, all of them are from Open Source projects.
[combined.zip](https://github.com/ThisIsNotRocketScience/GerberTools/files/6098389/combined.zip)
[Arduino_Micro.zip](https://github.com/ThisIsNotRocketScience/GerberTools/files/6098391/Arduino_Micro.zip)
[StepStick_Elecrow.zip](https://github.com/ThisIsNotRocketScience/GerberTools/files/6098393/StepStick_Elecrow.zip)
[V0_Display.zip](https://github.com/ThisIsNotRocketScience/GerberTools/files/6098397/V0_Display.zip)

Gautamjhawar, 2021-03-10:
> 
> 
> > Try using https://github.com/StanislavJochman/GerberTools_Eagle_Fix
> 
> If the issue were only about proper file names, I would have solved that myself long ago. I've been renaming manually some files, specially drill files from KiCad. Everything works, _except_ both silkscreen files. This is a bug, and a very annoying one, since there is no open source tool for Gerber editing so I could manually place the silkscreen on the combined board.

can you attach a picture of the starting pcbs and the merged output? i have been using it with overflowing silkscreen  and it worked just fine.

StijnKuipers, 2021-03-18:
Have you tried disabling the "Clip to board outlines" option in the panel
properties?

robstoddard, 2021-11-27:
I have run into my entire bottom silkscreen disappearing, and when I tried "Clip to board outlines" I get the following result on the bottom (top is great).   Project files attached.  Using current Eagle for board design.
[image]
[toStjin.zip](https://github.com/ThisIsNotRocketScience/GerberTools/files/7611924/toStjin.zip)
.

ckemere, 2021-11-27:
I've discovered that if there is any silkscreen on a layer outside the board outline, the whole silkscreen will be dropped. So I'm usually very careful during board design to make sure I'm not missing any small edges or whatnot.

CRImier, 2021-11-27:
@ckemere that's disable-able when you disable the "Clip to board outlines" option in panel settings.