ThisIsNotRocketScience/GerberTools #130: Unable to compile on Debian Buster

dansmolik, 2021-02-05:
Hi,
I try build GerberTools but I get this:

tmp/GerberTools-master/GerberProjects/GerberProjects.sln:  warning : Don't know how to handle GlobalSection ExtensibilityGlobals, Ignoring.
/tmp/GerberTools-master/GerberProjects/GerberProjects.sln:  warning : Don't know how to handle GlobalSection Performance, Ignoring.

Errors:

GerberProjects/GerberProjects.sln: error : Input string was not in a correct format.

	 41 Warning(s)
	 1 Error(s)

Time Elapsed 00:00:00.1746280

Thanks for help.

dayhawk4, 2021-03-24:
Hello,  using Debian as well and experiencing the exact same issue.  First note, in debian, sh is linked to dash, and so in order to get the install script to run correctly, enter as follows:
#bash ./install.sh
This will get around -o as an illegal option error
However, after this point, I find myself in the same boat as this thread except I am getting 14 warnings and 93 errors.

Went ahead and attached output I received as a text document including both standard out and error out.  Currently stuck as well.  
[CompileOutputOnBuster.txt](https://github.com/ThisIsNotRocketScience/GerberTools/files/6194905/CompileOutputOnBuster.txt)

dayhawk4, 2021-03-24:
Tried running "sudo bash ./install.sh".  Same results