ThisIsNotRocketScience/GerberTools #133: Issues parsing file with aperture  macros

luni64, 2021-03-03:
I'm using your library in the front end of an experimental solder paste dispenser. Generally it works very well but today  I stumbled over an issue with a Gerber file containing aperture macros. The library throws an exception during parsing the file. 

To exclude any issues with my application, I tried the file in your "TINRS Gerber Viewer" example which also throws an exception (see below).  

Here the problematic file [MM32-Bottom.zip](https://github.com/ThisIsNotRocketScience/GerberTools/files/6077729/MM32-Bottom.zip) which was  generated by KiCad. It works if I disable generation of aperture macros in KiCad.  The parser stumbles over these two macro definitions (line 15ff)

```
%AMFreePoly0*
4,1,22,0.550000,-0.750000,0.000000,-0.750000,0.000000,-0.745033,-0.079941,-0.743568,-0.215256,-0.701293,-0.333266,-0.622738,-0.424486,-0.514219,-0.481581,-0.384460,-0.499164,-0.250000,-0.500000,-0.250000,-0.500000,0.250000,-0.499164,0.250000,-0.499963,0.256109,-0.478152,0.396186,-0.417904,0.524511,-0.324060,0.630769,-0.204165,0.706417,-0.067858,0.745374,0.000000,0.744959,0.000000,0.750000,
0.550000,0.750000,0.550000,-0.750000,0.550000,-0.750000,$1*%
%AMFreePoly1*
4,1,20,0.000000,0.744959,0.073905,0.744508,0.209726,0.703889,0.328688,0.626782,0.421226,0.519385,0.479903,0.390333,0.500000,0.250000,0.500000,-0.250000,0.499851,-0.262216,0.476331,-0.402017,0.414519,-0.529596,0.319384,-0.634700,0.198574,-0.708877,0.061801,-0.746166,0.000000,-0.745033,0.000000,-0.750000,-0.550000,-0.750000,-0.550000,0.750000,0.000000,0.750000,0.000000,0.744959,
0.000000,0.744959,$1*%
```
Specifically it tries to parse the $1 parameter definitions as double which fails (see line 1040ff in gerber.cs).

Here the error message of your Gerber viewer:

[image]

StijnKuipers, 2021-03-18:
[image]

this is generated with latest version of the code.. is this correct? (no more crashing at least in the file you sent in the zip)

luni64, 2021-03-19:
Looks good. I'll test it in my application on the weekend. Thanks for fixing this.