ThisIsNotRocketScience/GerberTools #134: Solder mask, silkscreen and solderpaste are missing after using panelizer tool

azrilhafizi, 2021-03-09:
I try to merge together a few board by using the panelizer  tool, but the solder mask, silkscreen and solderpaste are missing from the output. The Gerber Viewer tool display the board correctly.
[image]
[image]

Output from Eagle
[image]

Output from Gerber Panelizer 
[image]

azrilhafizi, 2021-03-09:
I figured it out. New version of Eagle changed the output format, so Panelizer would not recognized it. Using https://github.com/StanislavJochman/GerberTools_Eagle_Fix I managed to fix it

closed 2021-03-09 by azrilhafizi

MayckonJhonatas, 2022-01-22:
hi, did you fixed it? i'm having the same problem.